siemens/ros-sharp #8: Fix Inertia Import in Unity3D

analogtwin, 2017-10-05:
UnityEngine.rigidbody specifies the Inertia of a rigid body via:
* Vector3 UnityEngine.rigidbody.inertiaTensor 
* Quaternion UnityEngine.rigidbody.inertiaTensorRotation
see: https://docs.unity3d.com/ScriptReference/Rigidbody-inertiaTensor.html

URDF specifies the inertia of a rigid body via:
* the 6 upper-diagonal  elements of the 3x3 inertia matrix 
i_xx, i_yy, i_zz, i_xy, i_xz, i_yz,
see: http://wiki.ros.org/urdf/Tutorials/Adding%20Physical%20and%20Collision%20Properties%20to%20a%20URDF%20Model

In order to import URDF's inertia matrix in Unity3D we have to:

1 diagonalize the matrix D = R*I*R'
* D is the diagonalized matrix
* R is a transformation matrix with norm 1, i.e. a rotation matrix
* I is the URDF inertia matrix
* R' is the transpose of R

2 fill Unity's inertiaTensor with the 3 diagonal elements of D

3 convert the rotation matrix in a quaternion and store it as inertiaTensorRation

We should try to use simple Linear Algebra and preferably no additional libraries, or at least very common math libraries.

MartinBischoff, 2017-11-02:
We are going to use Math.NET numerics. 
Evd is what we need:
https://numerics.mathdotnet.com/api/MathNet.Numerics.LinearAlgebra.Factorization/Evd%601.htm

MartinBischoff, 2017-11-13:
done in commit https://github.com/siemens/ros-sharp/commit/3efae53d6740b7d7d26ad947616abde9fe084869

In addition to calculating the inertia info required by Unitry3D we needed to ensure that this data is acutally used.

Whenever colliders are added to a rigidbody gameobject or its children
[Rigidbody.ResetCenterOFMass()](https://docs.unity3d.com/ScriptReference/Rigidbody.ResetCenterOfMass.html) and 
[Rigidbody.ResetInertiaTensor()](https://docs.unity3d.com/ScriptReference/Rigidbody.ResetInertiaTensor.html) are called.

We use [RigidbodyData.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RigidbodyData.cs) to maintain the original center of mass and inertia apecified in the URDF, or alternatively, to let Unity3D recalculate them.

closed 2017-11-13 by MartinBischoff