siemens/ros-sharp #15: Problems while trying to run Getting Started 2 tutorial

lucascoelhof, 2017-11-01:
Hi all,

First of all, thanks for providing this ROS package, it's been very helpful so far. I am also using ROS in combination with Unity for my masters thesis. 

When I run your tutorial [Getting Started 2](https://github.com/siemens/ros-sharp/wiki/GettingStarted2), I'm facing two problems.

First, the ros-sharp/UrdfImporter/UrdfSharpUnity3D folder mentioned during "Setting up Unity" section does not exist anymore. But I was able to get everything working by copying and pasting the  Unity3D/Assets/RosSharp folder on the Assets folder of my project. Should we change the wiki? Also, I noticed that there is a unitypackage file inside Release folder. I included this file and I was able to successfully see the URDF Model in 3D Object menu.

And second, and more critical, I could successfully import the model of the robot in Unity, but when I clicked on play, the client does not connect and subscribe to the robot model topics. So, I can see my robot moving in Rviz when I click on randomize, but I cannot see it moving in Unity. Any tips on why this is not working as expected?

MartinBischoff, 2017-11-01:
Hi Lucas,

thank you for using ROS# and getting in touch with us!

We are currently restructuring the Unity3D Tutorial project and, correspondingly, the Wiki page.
So yes, the Getting Started 2 Wiki Page is currently outdated and we should change it! This task is already adressed in Issue https://github.com/siemens/ros-sharp/issues/13.

The second Problem you have is also because we have not described the new workflow in the wiki yet.
The Urdf importer imports the plain Urdf model, and does nothing more. In order to use ros JointState topics to animate the model you have to add the following components:
1 add one single RosConnector to establish a connection to ROS.
2 add one single JointStateSubscriber in the same GameObject as the RosConnector.
3 add one JointStateManager in each HingeJoint of your Urdf Model. This script updates the transforms of your model based on the data recieved by JointStateSubscriber.
We implemented a JointStatePatcher to automate  Task 3 for us.
Also make sure all rigidbodies of your model are kinematic when animating your model via ROS. We also have a corresponding Patcher for this.

I hope this info helps you until we can provide an updated Tutorial project and updated Wiki pages.

Any help is highly appreciated.

lucascoelhof, 2017-11-01:
It worked! You guys are awesome!!!

I will try to use PR2 model on this and later on a group of robots. I will keep you guys updated with my progress and I will pull my examples for this repo as well, once I got them running.

closed 2017-11-01 by lucascoelhof

MartinBischoff, 2017-11-01:
Im glad to hear these good news!

Please do keep me updated. I'm looking forward to hear more about your application.