siemens/ros-sharp #16: Cloning rosbridge does not install all dependencies

lucascoelhof, 2017-11-03:
Hi all,

I know you guys are improving the documentation, but I would like to give a heads up about the current steps.

At least for me, using rosbridge by cloning its repo didn't work. Even after compiling it. I got this error while running the launch file:

Traceback (most recent call last):
  File "/home/lucas/catkin_ws/src/rosbridge_suite/rosapi/scripts/rosapi_node", line 36, in <module>
    from rosapi import proxy, objectutils, params
ImportError: No module named rosapi
Traceback (most recent call last):
  File "/home/lucas/catkin_ws/src/rosbridge_suite/rosbridge_server/scripts/rosbridge_websocket", line 39, in <module>
    from tornado.ioloop import IOLoop
ImportError: No module named tornado.ioloop

and then rosbridge node crashed.

However, installing rosbridge like described in their documentation http://wiki.ros.org/rosbridge_suite works like a charm. 

```sudo apt-get install ros-kinetic-rosbridge-server ```

MartinBischoff, 2017-11-03:
Thank you for this feedback. We will consider it when updating the wiki.
In my ROS environment I acutally installed rosbridge_suite as you did it.

MartinBischoff, 2017-11-03:
As this is a minor change., I've updated the Wiki right now.

closed 2017-11-03 by MartinBischoff

kauffmanes, 2018-06-07:
I just cloned it and also ran into this issue (maybe not for the same reason as above). Running `rosdep install --from-paths src --ignore-src -r -y` from the top of the workspace worked for me.