siemens/ros-sharp #18: my links seperated with UrdfImporter

Gongkaka, 2017-11-08:
When I try to import my own robot URDF file into Unity, all the links are separated. This may be due to the wrong transformation between links. However, the model works well after MoveIt! setup and interacting in RViz...any reasons that may cause that issue?

MartinBischoff, 2017-11-08:
Hello Gongkaka,
thank you for working with ROS#! What do you mean with "all the links are separated"? Please send us a clear description and 
if helpful screenshot(s).
Best regards,
Martin

Gongkaka, 2017-11-08:
Hi Martin,
Thanks for your reply! Below is the screenshot of my robot model imported from ROS to Unity.
[image]
Those links are not in the correct position among each other...but the transformations are already defined in URDF file and works well in RViz..

MartinBischoff, 2017-11-08:
Thmks for the screenshot. Please do the following:

1. Please make sure that your robot model is strictly according to the URDF format as specified here: http://wiki.ros.org/urdf/XML. See the link section in particular.

2. Please make sure that you included the complete RosSharp.unitypackage in your Asset folder https://github.com/siemens/ros-sharp/blob/master/Release/RosSharp.unitypackage.
If not, please delete your robot model data from the Unity Assets Folder, import RosSharp.unitypackage (select all files for import) and re-import the robot.

3. If this does not help, could you please send me your URDF+resource files, so that I can reproduce this error?

Did I undertand you right that the model looks correct in Unity3d after moveit setup and rviz interaction?
Which mesh file type are you using? Are you using any unconventional mesh orientation?
Have you tried importing some other ROS robot?

Gongkaka, 2017-11-09:
Hi,
Following your solution tips:
1. URDF format is correct as it works well in rviz interaction.
2. I tried again including the complete RosSharp.unitypackage in my Asset folder but the issue still exists.
3. Attached is my URDF resouce files I'm using. I am really appreciate if you can help inspect the error..
[abb_irb140.zip](https://github.com/siemens/ros-sharp/files/1456396/abb_irb140.zip)

MartinBischoff, 2017-11-09:
Hi Gongkaka,

thanks for providing the urdf and the meshes.
I can confirm the error on my side and that this is a valid issue as it is a bug in the URDF importer.

Though I have no solution yet, I can explain the problem:

In `ColladaAssetPostProcessor.cs` we specify how the Collada meshes will be oriented in Unity3D.
Unity3D is using a left-handed coordinate system with Y up
Collada meshes, by contrast, are based on right-handed coordinate systems. The upward-axis is specified in the .dae file itself in the xml tag `<up_axis>`. In your files we have  `Z_UP `, i.e. Z-axis pointing upwards.

Thus, for your files a rotation in line 63 is applied, which is obviously wrong.
We need some coordinate transformation similar to that specified in method `UnityCoordTrafo` 
in file `StlImporter.cs`.

Some of your meshes additionally have an initial offset, which must also be transformed into Unity3d coordiante system but currently is taken as is.
Additionally gripper opened2 has multiple child meshes. We must ensure that the transformation is not only applied to the parent object of the mesh, but also all child gameobjects.

Though I have not tired it myself I presume that one workaround for this version is to export the meshes as stl files in blender and use those in the urdf. 

I am looking forward to personally resolve this bug.
As I will probably not find time in the next two weeks, I ask the community for support.

MartinBischoff, 2017-11-14:
Hi Gongkaka,
commit https://github.com/siemens/ros-sharp/commit/4919380e6d1491dc42c5f8d49729d243d5c6d2a0 provides a new RosSharp.unitypackage with fixes for your robot.

The prroblem was that the Collada files linked in the URDF have rather complex mesh structures, some  with child sub-meshes, some with offsets and/or rotations.

I ended up using this urdf 
[abb_irb140_mb.zip](https://github.com/siemens/ros-sharp/files/1471563/abb_irb140_mb.zip)
 which corresponds to your URDF file `abb_irb140_mb` with the difference that I included the origins of the colliders as given in your urdf `abb_irb140_collision.urdf`.

Here is a screenshot of the result:
[image]

Note that you have to re-import all meshes in your asset folder for the orientation fixes to take effect.

As discussed in issue #19 you also have not defined any inertial in your URDF. Accordingly now Rigidbody and Joint Components will be created in Unity3D.

I hope these fixes work on your side as they do on mine.
Please let me know if you have any additional problems.

I'd be glad if  you can share your complete urdf model with us.

closed 2017-11-14 by MartinBischoff