siemens/ros-sharp #19: UrdfJointExtensions not being called to augment GameObject names with joint type

jaguar243, 2017-11-09:
Hi,

I am using the release version of the RosSharp unity package from `Release/RosSharp.unitypackage` to import `.urdf` files from a running instance of ROS. I'm able to import without any issues except that the GameObject names do not get augmented with the joint types which must be handled by the `RosSharp/Editor/UrdfImporter/UrdfJointExtensions.cs` script. I have checked my urdf file and have no other issues such as displaying it in rviz. I can't seem to figure out what I'm doing different, or maybe I'm missing something. 

Thanks!

MartinBischoff, 2017-11-09:
Hi Rahul!

I guess that for your applications you do not need an inertia as you do not want to simulate your robot in Unity but animate it via ROS.
That is the reason why you do not bother with specifing inertials in your URDF, right? ;-)

In Unity a joint requires a rigidbody. A rigidbody requires a mass. 
In URDF the mass of a link is specified in the inertial tag. In case no inertial exists, we can not specify the mass of the rigidbody. Thus we decided to better not add it. Therefore we can not add a joint.

Does this  make sense?

In case I guessed right and you have no inertials in your URDF, you have two options:

1 add dummy inertials in all you links in your urdf:

```
<inertial>
<origin xyz="0 0 0" rpy="0 0 0"/>
<mass value="1"/>
</inertial>
```
2 change the URDF importer behaviour: 
In class `UrdfLinkExtensions.cs` lines 32ff change

```
            if (link.inertial != null)
            {
                link.inertial.Create(gameObject);
                if (joint != null)
                    joint.Create(gameObject, parent);
            }
```
to 
 ```
            if (link.inertial != null)
                link.inertial.Create(gameObject);
            else
                gameObject.AddComponent<Rigidbody>();

             if (joint != null)
                    joint.Create(gameObject, parent);

```
Then a rigidbody of mass 1 will be generated if no inertal is specified. No matter what the acutal mass of that link is in reality. Above solution works well for the [R2D2 Urdf example](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch), I just tried it out.

Probably the most userfriendly solution will be to give a warning during the import if any joint component is left out due to a missing inertial.

What do you think?

jaguar243, 2017-11-09:
Hi Martin,
That makes a lot of sense! Thank you. I ended up using a default inertial
macro for all my links and it works fine now. A warning will be nice but in
general, I guess people will define inertial elements for the links for
actual robot control using ROS. Thank you for such a quick response!

Best,
Rahul

MartinBischoff, 2017-11-10:
We will add a Debug.Warning in `UrdfLinkExtensions.cs` whenever a
Joint component is skippedin Unity3D since the Inertial is not specified.

This warning is necessary since Inertials are optional in URDF.
See http://wiki.ros.org/urdf/XML/link

MartinBischoff, 2017-11-13:
done in commit https://github.com/siemens/ros-sharp/commit/897cc76c4567d1ff2633e84b59f19e7846989e09.

closed 2017-11-13 by MartinBischoff