siemens/ros-sharp #21: Which .NET Platform should be supported by UrdfImporter and RosBridgeClient?

MartinBischoff, 2017-11-24:
We developed ROS# primarily for communicating with ROS from within Unity3D.
Official Unity3D Versions still support .NET Framework 3.5 only.

For Non-Unity3D-Applications of ROS#, UrdfImporter and RosBridgeClient could be provided in a more recent, "standardized" .NET platform. 
Currently discussed versions are:
* .NET Core for RosBridgeClient
* .NEt Standard 2.0 for UrdfImporter
See https://docs.microsoft.com/en-us/dotnet/standard/ for an overview on .NET platforms.

Please use this issue to let us know the platforms you requrire for your project. 
Please support us by providing suggestions how to manage multiple Ppatform releases in one Visual Studio Project as upgrading from .NET Framework 3.5 would complicate the applicabilty in Unity3D.

@blommers

jaguar243, 2017-11-25:
To be able to use ROS# with unity and hololens, the project mist be built in the Universal Windows Platform (UWP), but the websocket library on which Rosbridgeclient is built on is not supported in UWP. So for UWP platform a custom websocket library needs to be built. I would suggest keeping the communication protocol user defined in Rosbridgeclient so that users can choose which protocol to use without too much change in the source code.

MartinBischoff, 2017-11-27:
From what I read [here](https://msdn.microsoft.com/magazine/mt590967.aspx) UWP also supports .NET Core Framework.

From .NET Framework 4.5 on, .NET has its own [WebSocket Class](https://msdn.microsoft.com/de-de/library/system.net.websockets.websocket) such that no external dependency as 
[websocket.sharp](https://github.com/sta/websocket-sharp) will be required. So our code will become even nicer and simpler when upgrading to .NET Standard 2.0.

Jasonthefirst, 2017-12-14:
Like suggested I'll give my two cents here, too. 
For my project it would be awesome if the ros-sharp library would support the hololens and therefore UWP Apps. I am not quite sure what the problem is and what needs to be done but there are additional projects at my university and especially at my institute working with the hololens connecting to robots via ROS. Right now unity has problems during compilation with the websocket and newtonsoft-json. 
Right now we are using an example project from h2r here on github to connect with ros. 
Thank you for your awesome project and making it open to use.

MartinBischoff, 2017-12-14:
@jaguar243 have you made progress with an UWP version of RosBridgeClient? Can you advise here?

jaguar243, 2017-12-14:
I'll commit my changes as a separate branch in the coming few days. Basically, I had to rebuild by RosBridgeClient using my custom websocket plugin that was compatible with UWP. No changes are needed on the ROS side.

MartinBischoff, 2018-01-09:
I found [this ](https://github.com/sta/websocket-sharp/pull/299) pull request on the websocket-sharp github repository stating that it provides a websocket-sharp version supporting  .NET Core.

Anyone interested in UWP support of ROS# interested in trying it out and report the results here?

Any suggestion on other websocket implementations to use instead of websocket-sharp?
@Jasonthefirst @arunavanag591 @jaguar243

MartinBischoff, 2018-05-11:
Thanks for sharing this info @dwhit . We've been following Unity's .NET platform support and waited for such a release supporting .NET 4.5+. It has been announced for such a long time already.

You are absolutely right: We can now also upgrade to .NET 4.7 and replace _websocket-sharp_ by the _.NET WebSocket class_ mentioned above.

Let us continue the discussion on this topic in #59.

closed 2018-05-11 by MartinBischoff

eisenWu, 2019-11-08:
Hi @MartinBischoff ,
i got the problem today, i am using websocket.sharp, but it dose not support UWP, as you posted above, it can be replaced by WebSocket Class, how it works, i am newbee to progromming by the way, but i will try to solve the problem, can you give me some tips?
Thanks in advance
Wei