siemens/ros-sharp #22: Which Publisher and Subscriber Examples should be provided in the ROS#/Unity3D project?

MartinBischoff, 2017-11-24:
ROS#/Unity3D currently provides the following Subscription / Publication Examples:

Subscriber-Examples for the Topics:
* `/joint_states` https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/JointStateSubscriber.cs
* `/odom`  https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/OdometrySubscriber.cs

We are planning to include more examples and to exdend above list.

On the one hand we want to keep ROS# generic and easy-to-learn and not to overload our library it with too much similar code. So users can get the hang of it by some examples and make their individual extensions and adjustments for their robot’s particularities.
On the other hand we want to have a preferably complete and ready-to use library.

Let us use this issue to discuss ideas which Subscription / Publication Examples are worth including.

MartinBischoff, 2017-11-24:
Subscription Ideas:
* `/map`  providing a `nav_msgs/OccupancyGrid` message for illustration in Unity3D
*  `/camera/` image data for illustration in Unity3D

Publication Ideas:
*` /cmd_vel_mux/input/teleop`  providing a `geometry_msgs/Twist` message for setting target velocities

lucascoelhof, 2017-11-25:
Hi all,

Concerning the subscription ideas, I think it's worth to invest in creating more message types that are missing. So far I've implemented some ones I needed, like Pose, PoseStamped and Image. 

Here are my ImageManager and ImageSubscriber classes:
https://github.com/lucascoelhof/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/ImageManager.cs
https://github.com/lucascoelhof/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/ImageSubscriber.cs
However I had some issues about the encodings, which I explained better here: https://github.com/lucascoelhof/ros-sharp/issues/1

I also implemented some Publication classes, but they need some refactoring and once I fix them I will post here. But here is a code snippet of a Pose publisher. 
```

[RequireComponent(typeof (RosConnector))]
public class PosePublisher: MonoBehaviour {
  private RosSocket rosSocket;
  public string topic = "/pose";
  private int advertizer;
  public GameObject UnityGameObject;
  private PoseTransformManager poseTransformManager;

  public void Start() {
    rosSocket = transform.GetComponent < RosConnector > ().RosSocket;
    advertizer = rosSocket.Advertize(topic, "geometry_msgs/Pose");
    if (UnityGameObject != null)
      poseTransformManager = UnityGameObject.GetComponent < PoseTransformManager > ();
  }

  public void Update() {
    this.publishPose(poseTransformManager.getPose());
  }

  public void publishPose(GeometryPose message) {
    rosSocket.Publish(advertizer, message);
  }

  public void setPublisher(string topic) {
    rosSocket.Unadvertize(advertizer);
    advertizer = rosSocket.Advertize(topic, "geometry_msgs/Pose");
  }
}
```

One thing I thought it was weird with the RosBridgeClient code is the name "advertize" with Z. Although advertize is correct English, ROS uses "advertise", with S. Here is an example: http://docs.ros.org/api/roscpp/html/classros_1_1NodeHandle.html#a6b655c04f32c4c967d49799ff9312ac6

jeremyfix, 2017-12-08:
I think the HelloWorld within the ROS community is the Turtlebot (see for example [http://wiki.ros.org/turtlebot_simulator](http://wiki.ros.org/turtlebot_simulator) )

Given you have a Unity scene with the URDF model of the turtlebot , I think it would make sense to have a basic scene with a floor, the turtlebot, a camera mounted on it, and adding the VelocityManager/Subscriber   CameraImage* , ROSConnector in the scene and controlling the simulation from  ROS nodes using standard teleop nodes such as turtlebot_keyboard_teleop

And doing so as on the Wiki with a kind of Tutorial explaining the click and drop for bringing the scripts, assigning the UrdfModels, ..

All the elements are already in ros-sharp and I think this is valuable for people wanting to use Unity as, say, a replacement of VREP, Gazebo, etc..

NikolasE, 2017-12-15:
A visualization of an RGBD-Pointcloud (as you already show in your videos) would be great.

MartinBischoff, 2017-12-18:
@NikolasE you are talking about [this video](https://www.youtube.com/watch?v=OytzagQirrk).
We decided not to upload these sbscriber, processor and manager scripts as they are extremely robot-specific, tuned for our hardware, and quite complex.
We fear that including such types of examples in ROS# will result many follow-up questions, frustration and no help for most people.

MartinBischoff, 2017-12-18:
@jeremyfix: We will consider your suggestions when revising the tutorial projects next time. I'm thinking about two contrasting scenes with the Turtlebot 2. In one scene control signals are published from Unity, and joint states, odemtry and a 2-d  rgb camera image is received. The robot is simulated in Gazebo on the ROS side. In the other scene the communication goes exactly the other way round and the robots physics are simulated in unity.

Adding the Subscribers & Publishers used in that tutorial scene should give a generic and comprehensive list of examples.

MartinBischoff, 2018-02-09:
Coding for the message types is done. See issue #23.

closed 2018-02-09 by MartinBischoff