siemens/ros-sharp #25: Communication with Ros bridge fails with built game

jeremyfix, 2017-12-07:
Unity Version : 2017.2.0f4
Platform : Linux Ubuntu 16.04

Below I describe an issue encountered when building the game and running the built game. This issue does not appear if the game is executed from the Unity Editor. 

Building the application for Linux x86_64 works. Then running the application, the websocket client successfully gets connected to the rosbridge node but there appears to be no message transiting on the topics and a warning is raised by the rosbridge node : 

[WARN] [1512640449.098013]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.

MartinBischoff, 2017-12-08:
Would activate  "Developmend Build" in the Build Settings, build and run the application again and then post the log file here?
Further info on [Log Files](https://docs.unity3d.com/Manual/LogFiles.html) and [Build Settings](https://docs.unity3d.com/Manual/BuildSettings.html).

MartinBischoff, 2017-12-08:
Without knowing the details yet, I presume it is because of the limited platform support of [websocket-sharp](https://github.com/sta/websocket-sharp). This issue might be related to issue #21 .

Please check if the following changes in Player Settings/Other Settings/Configuration have an effect:
* switching API Compatibility Level from ".NET 2.0 Subset" to ".NET 2.0"
* switching Scripting Runtime Version from "Stable (.NET 3.5 Equivalent)" to "Experimental (.NET 4.6 Equivalent)"

jeremyfix, 2017-12-08:
I built a developpement build and below is the log. I do not really anything very informative from the log ..

Running the binary :
```
fix_jer@fixjer-XPS-12-9Q23:/opt/UnityProjects/RosMine/Builds⟫ ./mineROS.x86_64 
Set current directory to /opt/UnityProjects/RosMine/Builds
Found path: /opt/UnityProjects/RosMine/Builds/mineROS.x86_64
Mono path[0] = '/opt/UnityProjects/RosMine/Builds/mineROS_Data/Managed'
Mono config path = '/opt/UnityProjects/RosMine/Builds/mineROS_Data/Mono/etc'
PlayerConnection initialized from /opt/UnityProjects/RosMine/Builds/mineROS_Data (debug = 0)
PlayerConnection initialized network socket : 0.0.0.0 55194
Multi-casting "[IP] 172.21.14.22 [Port] 55194 [Flags] 2 [Guid] 96164194 [EditorId] 881713867 [Version] 1048832 [Id] LinuxPlayer(172.21.14.22) [Debug] 0" to [225.0.0.222:54997]...
Started listening to [0.0.0.0:55194]
Preloaded 'ScreenSelector.so'
PlayerConnection already initialized - listening to [0.0.0.0:55194]
Player data archive not found at `/opt/UnityProjects/RosMine/Builds/mineROS_Data/data.unity3d`, using local filesystem
Logging to /home/fix_jer/.config/unity3d/DefaultCompany/RosMine/Player.log
```

And the log

```
Desktop is 1920 x 1080 @ 59 Hz
New context 0x42d6830 created with attributes:
Initialize engine version: 2017.2.0f3 (ee86734cf592)
GfxDevice: creating device client; threaded=1
Renderer: Mesa DRI Intel(R) Ivybridge Mobile 
Vendor:   Intel Open Source Technology Center
Version:  4.2 (Core Profile) Mesa 17.4.0-devel
GLES:     0
 GL_3DFX_texture_compression_FXT1 GL_AMD_conservative_depth GL_AMD_draw_buffers_blend GL_AMD_seamless_cubemap_per_texture GL_AMD_shader_trinary_minmax GL_AMD_vertex_shader_layer GL_AMD_vertex_shader_viewport_index GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_APPLE_object_purgeable GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_compressed_texture_pixel_storage GL_ARB_compute_shader GL_ARB_conditional_render_inverted GL_ARB_conservative_depth GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_draw_indirect GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_lo
cation GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_shader GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_occlusion_query2 GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_robustness GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_d
raw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_420pack GL_ARB_shading_language_packing GL_ARB_sync GL_ARB_tessellation_shader GL_ARB_texture_barrier GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map_array GL_ARB_texture_filter_anisotropic GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_multisample GL_ARB_texture_non_power_of_two GL_ARB_texture_query_levels GL_ARB_texture_query_lod GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_texture_rgb10_a2ui GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_texture_swizzle GL_ARB_texture_view GL_ARB_timer_que
ry GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_transform_feedback_instanced GL_ARB_transform_feedback_overflow_query GL_ARB_uniform_buffer_object GL_ARB_vertex_array_bgra GL_ARB_vertex_array_object GL_ARB_vertex_attrib_64bit GL_ARB_vertex_attrib_binding GL_ARB_vertex_shader GL_ARB_vertex_type_10f_11f_11f_rev GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ATI_blend_equation_separate GL_ATI_texture_float GL_EXT_abgr GL_EXT_blend_equation_separate GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_sRGB GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_pixel_buffer_object GL_EXT_polygon_offset_clamp GL_EXT_provoking_vertex GL_EXT_shader_integer_mix GL_EXT_shader_samples_identical GL_EXT_texture_array GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_comp [... cut ...]
Connected to RosBridge: ws://172.21.14.22:9090
UnityEngine.DebugLogHandler:Internal_Log(LogType, String, Object)
UnityEngine.DebugLogHandler:LogFormat(LogType, Object, String, Object[])
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RosConnector:Awake() (at /opt/UnityProjects/RosMine/Assets/RosSharp/Scripts/RosConnector.cs:32)
 
(Filename: /opt/UnityProjects/RosMine/Assets/RosSharp/Scripts/RosConnector.cs Line: 32)
```

I will make a test following your suggestions in your second comment.

jeremyfix, 2017-12-08:
Good news, as you suggested :

* switching API Compatibility Level from ".NET 2.0 Subset" to ".NET 2.0"

solved the problem. Thank you !

closed 2017-12-08 by MartinBischoff

hmzzzzzz, 2019-04-02:
thanks, I solved my problem.