siemens/ros-sharp #28: Unity3D Camera Feed to ROS Topic /image.

pushkalkatara, 2017-12-14:
We get Texture2Ds in Unity which can be converted to Jpeg or Pngs and further base64 encodings. Can these images be transferred over RosBridge to convert it into an image topic.

MartinBischoff, 2017-12-14:
Hi @pushkalkatara ,

yes! Image data be converted into an image topic and sent via RosBridge.

Even better:
Since recently, we have a CameraImagePublisher as an example ROS#/Unity3D Component which (I think) does exactly what you are looking for. Please take a look at:
* https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImagePublisher.cs
* https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImageManager.cs

@jeremyfix

jeremyfix, 2017-12-14:
You are right. This is exactly the interest of the CameraImagexxxx . You simply drop them un your scene and it will publish you camera image as a CompressedImage topic on rosbridge. You can also adjust (even dynamically while the scene IS running) the compression rate.

MartinBischoff, 2017-12-16:
I'm closing this issue as I think the question is sufficiently answered and there were no follow-up comments in 2 days.

closed 2017-12-16 by MartinBischoff

pushkalkatara, 2017-12-17:
@MartinBischoff Sorry for the late reply. I have used this library https://github.com/michaeljenkin/unityros to send CompressedImage message in jpeg format, but it seemed to sink the rosbridge and video was lagging on the client side. Can you please help me with the fps.

Regards
Pushkal.

jeremyfix, 2017-12-17:
Please give a try with the CameraPublisher we provide.

Also, you might consider building (compiling) the game. On a small scene, as far as I remember, I got 2x more FPS with the built game.

best;

pushkalkatara, 2017-12-18:
Thanks for the help :+1:

zainmehdi, 2018-01-11:
@pushkalkatara Can you please help me out with the issue ? I attached the camera publisher script to my game object and configured ros but when I run the game there are no images being published. Would be thankful for help