siemens/ros-sharp #29: Wrong paths when importing URDF model in linux

jeremyfix, 2017-12-17:
I tried to import the Nextageopen URDF in Unity ; 

The URDF is correctly imported  but actually copied in an inappropriate path ; 

My project is named DroneControl and, instead of copying the assets in   DroneControl/Assets/Urdf  , it copied it into     DroneControl\Assets\Urdf       and therefore the object does not appear in your project in Unity , because "DroneControl\Assets\Urdf" is not a subdirectory of DroneControl.

If I copy the assets in the appropriate path , the model is correctly displayed within Unity. So , it is just a question of "/" and "\" in the path of the UrdfImporter (I suppose) ;

MartinBischoff, 2017-12-18:
Thank you for this info. To be honest, we never tested ROS# on any other OS but Windows 7 and 10.
From what I read in your reports, it seems like RosSharp/RosBridgeClient works fine on Linux.
Given that you can even import Robot with RosSharp/UrdfImporter in Linux and only the assets paths incorrect, we are already quite close to a version of ROS# which equally works on Winows and Linux.

Here are some hints to a solution to the problem you described:
Likely, we need to make some adjustments with `Path.DirectorySeparator` and  `Path.AltDirectorySeparator` in the following files:
* [RosBridgeClient/UrdfImporter.cs](https://github.com/siemens/ros-sharp/blob/master/RosBridgeClient/UrdfImporter.cs), method `getLocalFilename`.
* [Unity3D/UrdfImporter](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/UrdfImporter/UrdfAssetDatabase.cs) mehtods  `GetAssetPathFromPackagePath` and`getMaterialAssetPath`.

Ideally, we find one solution that works on both OSes.
If not, we can distinguish between different OSs  for individual lines of code. 
* in the RosBridgeClient project as described [here](https://softwarerecs.stackexchange.com/questions/13717/tiny-c-library-to-tell-whether-windows-mac-linux-android-ios).
* in Unity as described [here](https://docs.unity3d.com/ScriptReference/Application-platform.html)

NikolasE, 2017-12-19:
Adaption is also needed in UrdfImporterEditorWindow.cs

jeremyfix, 2017-12-30:
Ahhhh, you are right, the problem actually came from the definition of the path in the UrdfImporterEditorWindow.cs

MartinBischoff, 2017-12-31:
Thank you @jeremyfix for fixing this bug!

closed 2017-12-31 by MartinBischoff