siemens/ros-sharp #34: Advertize -> Advertise

peci1, 2018-01-09:
Compare: 
https://github.com/siemens/ros-sharp/blob/1423fbb93d42f602b36446a5dbd545224812924b/RosBridgeClient/RosSocket.cs#L53

https://github.com/ros/ros_comm/blob/lunar-devel/clients/roscpp/src/libros/node_handle.cpp#L285

MartinBischoff, 2018-01-10:
Hi @peci1!
Thank you for your interest in ROS# and for spell-checking our code!

This is no typo, we named that method Advertize on purpose. Our objective is to integrate ROS in a best-possible manner in the .NET / C# / Unity world. In this context we are trying to stick to the [Coding Style](https://msdn.microsoft.com/en-us/library/ms229043%28v=vs.100%29.aspx) used in these environments and names do not match the original naming in ROS anyway and we do not take cpp code as a reference. As names are different anyway, we took the liberty and named the method to what we thought is best American English.

Triggered by your question and googling I found out that advertise is not only correct in American English as well, it also seems to be more common than advertize. We are thus thinking about following your suggestion and renaming the method. At least this will less likely look like a spelling mistake and reduce change reqeusts in the future.

Any futher opionion about this issue? Native American English speakers?

MartinBischoff, 2018-03-20:
Convinced. Advertise is correct english. Renamed in commit https://github.com/siemens/ros-sharp/commit/923fcd630cdc3edf9fb08edd98c7c422abb42dde.

closed 2018-03-20 by MartinBischoff