siemens/ros-sharp #35: Simulation

peci1, 2018-01-09:
I've just read the ROS# announcement, and saw that you use URDF to simulate robots. This format is not suitable for description of simulation environments. The better format to use is http://sdformat.org/ used by Gazebo. It allows for other important specifications like joint limits, friction coefficients etc... And for those left only with a URDF model, there is the `gzsdf` tool that converts URDF to SDF (which, of course, doesn't contain any of the simulation-specific tags).

But anyways, great work!

MartinBischoff, 2018-01-10:
Hi @peci1 !
Thank you for reading our blog post, your interest in ROS# and for finding our work great!

Thank you also for suggesting a new SDF import feature!
We are aware of the SDF format. In our department we are using URDF files only to specify our robot models. This is the reason why we opted for a URDF Importer (in the first place).

Please note that the URDF file is made for _simulation models_ of _individual robots_. It contains all relevant model parameters for rigidbody simulation models. Among others, also joint limits and friction coefficients can be specified in a URDF (http://wiki.ros.org/urdf/XML/joint).

To be honest, we are very glad that the URDF importer works so nicely in Unity and we found a Unity counterpart to (almost) all specifications can be made in an URDF.

The SDF importer specifies whole _simulation scenes_ with optionally _multiple robots_ in it. The sdf data format can thus be considered as a superset of the URDF data format. Though the specification of an  individual robots in an SDF does not match the specification of the same robot in a URDF, many things are identical.

An SDF importer in ROS# could be considered as a complete Gazebo-Unity Scene Importer.

When thinking about implementing an SdfImporter in addition to UrdfImporter in ROS#, I noticed the following:
* The whole UrdfImporter architecture can be adopted by the SdfImporter:
   * Step 1: Implement an xml file parser in a separate Visual Studio project to be used for not-necessarily-Unity applications. (The easy part).
  * Step 2: Use the dll obtained in step 1 in  Unity and implement C# extension methods in for interpreting the xmldata in Unity.
* A huge part of the UrdfImporter code can be re-used or shared with the SdfImporter:
   * The overall link & joint tree structure of robot models seems to be identical.
   * The mesh importers can directly be re-used.
   * many XML elements are identical, e.g. interal and interta elements which cause me some headache to convert to a Unity [InertiaTensor](https://docs.unity3d.com/ScriptReference/Rigidbody-inertiaTensor.html)
* Some SDF-specific parameters are simple (e.g. the gravity value specified in <world><gravity> can directly be written in the Physics.gravity vector). Others will become complex (possibly skysphere).

Rather than an SdfImporter, I see the biggest potential in an SdfExporter: Think about setting up a whole Gazebo Simulation Environment with the efficient Unity Editor functionalities and exporting that scene to ROS.

peci1, 2018-01-10:
You're right that the formats are similar, but I'd still refrain from telling that by importing URDF you can start a simulation. Most of the required parameters can be described in both URDF and SDF, but in a URDF you can't e.g. set a link as self-colliding, kinematic-only etc. Or, you can, but inside the `<gazebo>` element, which is however exactly intended to be used during the translation to SDF. I'm also using this approach.

To sum it up - the URDF import is a good start, but maybe you should warn a little that it may not cover all the simulation description needed to simulate some models (probably it's good enough for everything you use).

And the SDF export looks really like a good idea ;)

MartinBischoff, 2018-09-21:
In [this](https://github.com/siemens/ros-sharp/commit/ab180797bc3b313fa4d9e34472d7469f073c826a)  commit we realized exporting URDFs from Unity.
Since we are not currently planning to further go into SDF im-/export I am closing this issue.

closed 2018-09-21 by MartinBischoff

rahuldeo2047, 2019-02-05:
Hi @MartinBischoff ,

Please allow me to share our views.
We came across this repository. We are very excited to evaluate this tool.

We understand why urdf is important. One reason we could find is that the ROS nav and similar packages expect URDF as robot description. However, one of the major limitation we came across while working with an URDF model is; we can not simulate those have chained joints. So we had to stick to sdf based models. chained joints models i.e. parallel gripper, complex suspension systems etc. Or the joints where parent and child links form a loop.

We think as the awareness for urdf limitation grow, the community might stick to sdf rather than that of URDF.  Hence, the request for having an SDF importer can be a valuable library for few developers like us. I wish to +1 for the need of sdf here.