siemens/ros-sharp #36: How to subscribe to custom messages using ros-sharp?

pushkalkatara, 2018-01-22:
Please help me how to define headers for custom messages and subscribe them in Unity.

MartinBischoff, 2018-01-23:
Hi @pushkalkatara,

Please proceed as follows:
* Add your custom message type in [message.cs](https://github.com/siemens/ros-sharp/blob/master/RosBridgeClient/Message.cs) or in a separate file. (It makes no difference, see below.)
* Build the RosBridgeClient Solution and replace the existing RosBridgeClient.dll in the Unity Asset Plugins folder.
* Write a _Subscriber_ script for your new Message. You can take e.g.[ OdometrySubscriber.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/OdometrySubscriber.cs) as a reference.
* Write a _Manager_ oder _Writer_ script to apply the message contents in your Unity scene. You can take [OdometryTransformManager.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/OdometryTransformManager.cs) as a reference.

For the compiler it makes no difference whether you put the message type classes in separate files or not. In [Message.cs](https://github.com/siemens/ros-sharp/blob/master/RosBridgeClient/Message.cs), by way of exception, we defined all message types in one file since we think we think it is easier to keep the overview. Of course you can also split up that file and/or put your new message types in separate files if you want to. As I said it makes no difference.

I hope this answers your question.

pushkalkatara, 2018-01-23:
Thanks for the help. You are amazing. :D

closed 2018-01-23 by pushkalkatara

liucong-roadstar, 2019-02-26:
Hi @MartinBischoff,
Thanks for your great answer!

I have another question, I'm now working on Ubuntu and trying to subscribe to custom messages using ros-sharp from within Unity. I'm wondering, is there a way to build RosBridgeClient solution on Ubuntu? BTW, I'm using vs code as my editor.

Thanks!

liucong-roadstar, 2019-03-05:
For anyone who's wondering the same question, I just found out that:
On a ubuntu system, if you have mono installed, you can just enter the directory where RosSharp.sln resides, and run in termial:
msbuild

And that should do the work.