siemens/ros-sharp #37: Unity returning Null Object

pushkalkatara, 2018-01-27:
Rosbridge status - Client Connected, Subscribed to /topic.
Also 'rostopic echo /topic' is giving the right output.
Under subscriber callback function, printing Message, parsing and printing Message, both are giving Null Object. Help, please.

MartinBischoff, 2018-01-28:
There can be many reasons for this error on the way from ROS to Unity. We likely need more info for being able help.

* Are there any warning or error messages on the `rosbridge_server/rosbridge_websocket` node?
* Do you sucessfully receive other messages from ROS?
* What type is your topic?
* Can you send us files so that we can reproduce the error?

I presume that you have a custom message type which [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) does not know how to serialze.
As a quick-fix you can in this case convert your message data to a known message type, e.g. [std_msgs/String](https://docs.ros.org/api/std_msgs/html/msg/String.html) and write a C# script  in Unity that interprets the string message to access the data.

Things you can also try out:
* Subscribe to your topic in a console application like [this](https://github.com/siemens/ros-sharp/wiki/Info_CodeExample). If you receive your message as a non-Null Object there, we know that it is a RosBridgeClient / Unity problem. If not, we know that it is a rosbridge_suite / RosBridgeClient problem.
* If there are no error or warning messages on the `rosbridge_server/rosbridge_websocket` side, you might need to add your message type to the message_objects Dictionary in [RosSocket.cs](https://github.com/siemens/ros-sharp/blob/master/RosBridgeClient/RosSocket.cs)

Please inform us on your progress in solving this problem.

pushkalkatara, 2018-01-28:
Thanks for the reply.
I tried the HelloWorld example, Publisher from Unity to ROS is working and i am receiving the output on ROS side, but  Debug.Log(message) in the callback function in Unity is not printing "World!" in the logs. What am i doing wrong? 
For more reference, here are my custom messages -
Ros Side:
1. Elements
```
uint32 part_id 
float32 x
float32 y
````
2. Data
```
Elements[] parts
```

3. Datas
```
Data[] datas
Header header
```
Unity Side:
```
public class Elements : Message
    {
        public int part_id;
        public float x;
        public float y;
        public BodyPartElm()
        {
            part_id = 0;
            x = 0f;
            y = 0f;
        }
    }

    public class Data : Message
    {
        public Elements[] parts;
        public Data()
        {
            parts = new Elements[0];
        }
    }

    public class Datas : Message
    {
        public Data[] datas;
        public StandardHeader header;
        public Datas()
        {
            datas = new Data[0];
            header = new StandardHeader();
        }
    }
```
The subscriber:
```
rosSocket.Subscribe("/answers", "AnswersData/all_data", updatedatas, UpdateTime);

```
I tried adding the message to the Dictionary of subscribers in RosSocket but it didn't work.

MartinBischoff, 2018-01-30:
As written above, I believe that [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) does not know how to serialize your new message type. You have two options:

1. Convert your message to a message type that rosbridge_suite knows (e.g. std_msgs/String).
2.  Teach rosbridge_suite how to serialize your message type.  If you want to go for this option, you will find help [here](http://wiki.ros.org/rosbridge_suite).

pushkalkatara, 2018-01-31:
Can you please elaborate how can I serialize the message type?

MartinBischoff, 2018-01-31:
I do not know this myself, sorry. As it is not a ROS# issue but a rosbridge_suite problem you are having here, I'd suggest to close this issue at this point and you ask for support at [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) or [ros answers](https://answers.ros.org/questions/).

MartinBischoff, 2018-02-23:
@pushkalkatara any progress in this problem you were having?

pushkalkatara, 2018-02-24:
Couldn't find a workaround for this :/

PengC98, 2018-04-23:
I have a similar question! I just use the Info_CodeExample but i find that i can only publish a topic and cannot get the "world" . I think there is something wrong with the websocket.onmessage()

MartinBischoff, 2018-04-24:
Hi @PengC98 ! Thanks for trying out the example!
* We send [std_msgs/String.msg](https://docs.ros.org/api/std_msgs/html/msg/String.html) in this example. So the problem you are having is not related to this issue about _custom message types_ . Better open a new issue next time.
* Triggered by your comment we revised [Info-CodeExample.md](https://github.com/siemens/ros-sharp/wiki/Info_CodeExample): We updated `Advertize` to `Advertise` (which was changed in [this recent commit](https://github.com/siemens/ros-sharp/commit/923fcd630cdc3edf9fb08edd98c7c422abb42dde)). Then  this example works as it should. Here is a screenshot of the console windows (Ubuntu and Windows):
[image]
* If the problem persists on your machine, I'd suggest to check your firewall and proxy settings.
* Please also note that `websocket.onmessage()`is a method of [websocket-sharp](https://github.com/sta/websocket-sharp), the Websocket protocol implementation we are using. You might want to look for help [there](https://github.com/sta/websocket-sharp).
* For further support please open a new issue and provide a detailed error description.

closed 2018-05-03 by MartinBischoff