siemens/ros-sharp #38: Clarification about the tutorials step 1.4

ani-rudh, 2018-01-28:
Hi Devs,

I am completely new to ROS and I am trying to follow the tutorials in your wiki to control a robot in Unity 3D via ROS. Until the stage where I have to build the file_server (tutorial part 1.3), I could follow. I am having issues in stage 1.4 and 3.1:

- especially, if the 'getting_started2.launch' file already exists or should I create one? 
- If I created one, in which ROS package's launch folder should I place it, I tried file_server/launch, but the launch command didn't work?
- should I place it in rtmros_nextage/rtmros_nextage/launch(create the launch folder?) ?

Also, should the ros-sharp main package be cloned to catkin_ws/src or just the file_server suffices?

If the questions come across as silly, please do consider that I am totally new to ROS and still trying to understand how the mechanism of packages, wrokspaces, etc. function!

Thanks in advance,
Anirudh

verena-roehrl, 2018-01-29:
Hi Anirudh,
great to hear that you are working with ROS#.
- you have to create the [getting_started2.launch](https://github.com/siemens/ros-sharp/blob/master/ROS/tutorials/launch/getting_started2.launch) and [file_server.launch](https://github.com/siemens/ros-sharp/blob/master/ROS/tutorials/launch/file_server.launch) file of [Step 1.4.1](https://github.com/siemens/ros-sharp/wiki/User_Inst_NextageOpen) in the launch folder of a package (/PATH_TO_YOUR_PACKAGE/launch/)
- if a certain roslaunch file was not found it is most likely that you didn't source your `setup.bash` file. This command helps:
```
$ source ~/PATH_TO_YOUR_CATKIN_WS/devel/setup.bash
```
- in principle you could place your launch files in the launch folder of any package you like, but I don't recommend to place it in these big downloaded packages (_never touch a running system_). In addition the functionality of the mentioned launch file is not really related to the launch files in rtmros_nextage/rtmros_nextage/, so just keep the content of your packages structured depending on its functionality.
- you don't need to clone the whole ros-sharp repository to your catkin_ws. Just cloning the file_server and creating the launch files in the launch folder is enough.

Please check and go through the official [ROS Tutorials](http://wiki.ros.org/ROS/Tutorials). These help you to understand the whole mechanism of ROS (starting with small pieces) better and answers some of your questions as well :).

I hope that helps,
Verena

ani-rudh, 2018-01-29:
Hello Verena,

It certainly helped! I am now able to launch the server and communicate
with it from the Unity pop-up window (URDF Importer) to import the meshes.
Nevertheless, The client from unity side is communicating with the file
server but disconnecting without performing the import or the creation of
the GO in the scene hierarchy. (At office)
On the other hand, in one of my other systems, I am not able to get the
bridged adapter to have an IP address at all. Have you a solution to this?
I read that it is a VirtualBox issue but there are no working solutions
that helped! (At home)

Many Thanks in advance for your help,
Best Regards,
Anirudh

MartinBischoff, 2018-01-30:
Dear @ani-rudh ,please make sure to thoroughly read the [Contribution Guidelines](https://github.com/siemens/ros-sharp/blob/master/CONTRIBUTING.md) before opening an issue. Before asking, please also spend some time and try to find the answer to your question yourself.  Solving probelms and and learning is the daily business of a programmer. All questions which are not directly related to ROS# are placed better [here](https://answers.ros.org/questions/) or [there](https://stackoverflow.com/).
Thank you for your understanding.

closed 2018-01-30 by MartinBischoff