siemens/ros-sharp #39: A few wiki fixes

mcevoyandy, 2018-02-04:
Fixed a few typos and added some missing steps. Link to [my fork](https://github.com/mcevoyandy/ros-sharp/wiki). 

* Section 1.4.1 - 'git clone' command missing '.git'
* Section 3.2.1 - Spelling, added steps to create the referenced GameObjects, formatting

Link I found useful for [merging wikis](https://gist.github.com/larrybotha/10650410):

MartinBischoff, 2018-02-04:
Thank you @mcevoyandy for these clarifications. I just pushed your changes to the upstream Wiki. @roehrlverena fyi

closed 2018-02-04 by MartinBischoff

m-talha, 2019-03-21:
Fixed the path for a couple of files in section 3.3:

`
    JoyAxisJointTransformWriter.cs (/PATH_TO_YOUR_UNITY_PROJECT/Assets/RosSharp/Scripts/RosBridgeClient/MessageHandling/)`
`JoyAxisInputPasser.cs (/PATH_TO_YOUR_UNITY_PROJECT/Assets/RosSharp/Scripts/RosBridgeClient/MessageHandling/)
`

Link to my [fork](https://github.com/m-talha/ros-sharp/wiki).

MartinBischoff, 2019-03-22:
Thanks for the info @m-talha and for providing the corrections.
I just updated [the whole wiki page](https://github.com/siemens/ros-sharp/wiki/User_App_NoROS_AnimateRobotModelInUnity3D).
I hope this tutorial becomes better understandable now.
Please do open a new issue for new topics in the future. New issues in closed topics often get lost.