siemens/ros-sharp #40: URDF Import

mcevoyandy, 2018-02-07:
It seems only the following folder structure is valid:
```
robot_description_dir
   urdf_dir
      myrobot.urdf
      meshes_dir
         link1.stl
         link2.stl
         etc
```

Packages with the following structure fail:
```
robot_description_dir
   urdf_dir
      myrobot.urdf
   meshes_dir
      link1.stl
      link2.stl
      etc
```
I've tested this with franka and ur5 and get the same result. I can only import when I change the structure of the directory. Was there a reason to follow this structure on the Unity side or could this be followed to allow for various structures of the robot_description directory?

MartinBischoff, 2018-02-07:
Hi @mcevoyandy,

* You a right, there is only one valid folder structure.
* The folder structure is defined  the URDF mesh attribute `filename="package://..."`.
UrdfImporter expects the meshes to be at the same relative folder location as they are on the ROS side.
* When importing the URDF from ROS via [UrdfImporter](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/CustomEditor/UrdfImporterEditorWindow.cs) all resources are automatically placed in their correct folder sturcutre.
* When importing the URDF from a local file, via [UrdfGenerator](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/CustomEditor/UrdfGeneratorEditorWindow.cs) you need to make sure that the resources folder structure corresponds to that on the ROS side.

Some background info:

In theory URDF resources could be defined as follows:

```
\link_1
     mesh_link.stl
\link_2
     mesh_link.stl
```
If we did not consider the folder structure and let Unity search for the assets by filename, it would fail in this example and use the same mesh for both links.

[UrdfAssetDatabase](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/UrdfImporter/UrdfAssetDatabase.cs) is responsible for mapping `package:\\...` destinations to Unity Asset folder destinations. See e.g. methods [CreateVisual and CreateCollider](
https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/UrdfImporter/UrdfLinkGeometryMeshExtensions.cs) asking `UrdfAssetDatabase` where to find a mesh.

If you need more flexibility for your URDF mesh folder structure in Unity you might want to  adjust [UrdfAssetDatabase](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/UrdfImporter/UrdfAssetDatabase.cs) to your needs and e.g. ask the user for a root URDF resources folder somewhere in the Unity Assets folder structure.
For keeping it simple, I think would not like that in my ROS# version, though.

mcevoyandy, 2018-02-07:
Hi @MartinBischoff 

From my testing it seems that the assets get saved in the proper folder structure according to `filename="package://...` but then the error occurs during retrieval if the meshes aren't in the same directory as the urdf file. So in the second example I posted, the folder structure is maintained, but Unity fails to find the meshes in that location and the mesh object is null.

MartinBischoff, 2018-02-08:
I got your point now. You are right, the urdf file location specifies the `package://` root folder counterpart in the Unity Asset folder structure. 

This is coded in `GetAssetPathFromPackagePath` in [UrdfAssetDatabase](
https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Editor/UrdfImporter/UrdfAssetDatabase.cs).

We did that for the following reason:

Whether copied manually or by calling 
```
rosparam get /robot_description
```
via rosbridge, once a URDF file is in a Unity asset folder we do not know the original URDF file location in the ROS package.

Often there doesn't even exist a corresponding URDF file in the ROS package. Instead, the /robot_description parameter (which actually gives out what should be the URDF file contents) is often assembled from multiple xacro files located in various different package folders.

Does this answer your question? Do you have any improvement suggestions?
Which folder flexibility would you like to have?

MartinBischoff, 2018-02-14:
@mcevoyandy any comment? can we close this issue?

mcevoyandy, 2018-02-14:
I'd like to take a better look at it but haven't had time yet. I would be inclined to leave it open. Fixing this issue will make this package more accessible.

mcevoyandy, 2018-02-21:
I did some initial testing and have a few robots coming in with no problems, but am not sure if this solution is the best approach:

changes I made: (tests not done over rosbridge yet)
1. `assetPath` was made to be the path to the robot description package instead of the path to the URDF file
2. In `GetAssetPathFromPackagePath` I ignore the package declaration `package://package_name` instead of just `package://`

Anyway, what are your thoughts on changing the paths so they're relative to the package? Then the URDF importer should work for any custom structure so long as they copy in the whole package into the Assets folder.

MartinBischoff, 2018-02-21:
Sure, the folder structure can be changed as you discussed above.
However, how do you set the value of `assetPath` in general?

As I tried to explain already in my answer above:
We do not know the package root folder in the Unity asset folder structure. We use the location of the urdf file to set it.

Considering your example, users might put your robot description package e.g. in
* `Assets/urdf/robot_description_dir`
* `Assets/robots/robot_description_dir`
* `Assets/robot_description_dir`
* `Assets/robot_description_dir/robot_description_dir`

How do you identify the package root folder in Unity? Especially when importing the URDF via rosbridge_suite?

mcevoyandy, 2018-02-22:
For my tests I've been adding the whole robot description package under assets so have been looking for the package.xml file, which I think takes care of the examples you listed. 

And I haven't gotten to the rosbridge_suite yet... will try to look at that this weekend. If I'm able to come up with something solid I'll submit a pull request.

MartinBischoff, 2018-02-23:
Thanks @mcevoyandy for your efforts in this issue.

Please note that I am not convinced by the suggested modifications due to the following reasons:

* Some description packages have a lot of additional data like python code, meshes in various formats and multiple robot model variants that are not required by the Urdf model in Unity. Coppying the whole robot description package will clutter the Unity asset folders with addtional files, increase the build time and can even cause errors.

* As already mentioned above, some description packages have no urdf file at all. The contents of the `/robot_description` param is assembled by xacro files at runtime. When importing the urdf via rosbridge, where in the package folder structure would you put the urdf file, that is generated ion the Windows side only, if not in the root folder?

* The urdf importer is designed to copy all neccessary data from the Ubuntu side, not more not less. It searches the urdf for referenced resources and then search the collada files for referenced texture images. Only those files are copied into the asset folder. This minimizes the effort for importing and asset handling and keeps the folder nice and clean. We also have no package.xml when importing via rosbridge.

I still think the current folder structure and import procedure is the most convenient one. The only thing we should improve is to inform the users to put the urdf in the root folder when copying the files manually.

MartinBischoff, 2018-03-29:
We now have an [updated Wiki page](https://github.com/siemens/ros-sharp/wiki/User_App_NoROS_ImportingURDFOnWindows) explaining the file-based URDF import in more detail, esp. where to put the `.urdf` file.

We have not identified any better solution for generic ROS packages in this issue.  The existing solution is pratical and now also well documented.  I therefore suggest to close the issue. @mcevoyandy : do you agree?

MartinBischoff, 2018-05-11:
@mcevoyandy : I reopened this issue  due to your pull request #57 .

In order to achieve more flexibility in the URDF asset folder organization, why don't we simply ask the user where the root package folder is, if we can not find it?

MartinBischoff, 2018-05-18:
Having thought about this issue, I'd suggest to simplify the modifications in pull request #57 as follows:
* If the first resource (= mesh or texture) is not found, ask the user where the package root folder in the Asset Folder is.
  -> This resolves import problems where the URDF is not in the package root folder.
* If this resource, or any other resource is not found during the import process, ask the user where this particular resource is.
 -> This resolves import problems where the resource folder structure does not correspond to the package folder structure.

We can use [OpenFolderPanel](https://docs.unity3d.com/ScriptReference/EditorUtility.OpenFolderPanel.html) and [OpenFilePanel](https://docs.unity3d.com/ScriptReference/EditorUtility.OpenFilePanel.html) for asking the use the required infos.

In each dialog, the user shall have the opportunity to cancel the whole import (as many dialogs might have to be closed if a folder structure is messed up).

MartinBischoff, 2018-07-27:
Thank you @SuzannahSmith for realizing my suggested solution above!
Changes merged in commit https://github.com/siemens/ros-sharp/commit/28df9ba48d49181caad91114b1760c70b9162254.

closed 2018-07-27 by MartinBischoff