siemens/ros-sharp #42: Unity Texture2D to ROS CompressedImage Message

pushkalkatara, 2018-02-12:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu 16.04
* My Unity Version is: 
* My ROS Distribution is: ROS Kinectic
* My Build Target Platform is: 

**Here is my bug description:**

When sending Texture2D from unity to ROS as SensorCompressedImage Message, rosbridgeserver produces an error.

Perform the following steps reproduce the bug:

https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImageManager.cs
https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImagePublisher.cs

Observed results:

[image]

Expected results:

Texture2D should be received as ROS Messages.

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2018-02-13:
Thanks @pushkalkatara for reporting this error. 
It is fixed in the recent commit https://github.com/siemens/ros-sharp/commit/da2af1f16e4406956dd31804eb72999d170e3d6a

The problem was that we missed adding a [JsonIgnore] in [Message.cs](ros-sharp/RosBridgeClient/Message.cs) to the property RosMessageType. Because of that, RosBridgeServer could not de-serialize any of our messages as it didn't know where to put RosMessageType.

Please check if this error is also resolved on your side with the recent commit.  If so, please close this issue.

pushkalkatara, 2018-02-13:
Thanks @MartinBischoff for the commit. The issue is resolved but the video on the ROS Client side is lagging.

closed 2018-02-13 by pushkalkatara