siemens/ros-sharp #43: Texture2D to CompressedImage ROS lagging on the client.

pushkalkatara, 2018-02-13:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu 16.04
* My Unity Version is: 
* My ROS Distribution is: ROS Kinectic
* My Build Target Platform is: 

**Here is my bug description:**

Texture2D is being published using these scripts:
https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImageManager.cs
https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImagePublisher.cs
But the FPS received on the Client from Unity is very low and the video is lagging.

Perform the following steps reproduce the bug:

Observed results:

[image]
Lag observed on the client side. Also the server is hosted on localhost and ethernet powered, so most probably not an issue of network.

Expected results:

No Lag.

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2018-02-14:
This is a jitter you see here in RViz since your ROS System (rosbridge_suite and/or rviz) receives too many image publications, more than it can handle. It even mixes the order images are illustrated in Rviz since there are arriving so fast.

Looking at [CameraImageManager](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/CameraImageManager.cs) we see that images are published every single frame in Unity.

Solutions: 
* Reduce the image message publication frequency in Unity
* (and/or give your ROS system more resources.)

You can reduce that e.g. by adding something like:
``` 
public int FrameSkip = 10;

[...]

private void Update()
{
   if (Time.frameCount % FrameStep == 0)
   {
    <trigger message publication here>
    }
}
[...]
```
then play with the `FrameSkip` number to optimzie the performance for your system.

In the new message handling framework (see issue #23 ) this parameter is available in the generic Unity publisher component.

MartinBischoff, 2018-02-23:
@pushkalkatara does the suggested solution solve your problem? If so, please close the issue.

pushkalkatara, 2018-02-24:
Yes. Thanks for the solution. Closing the issue.

closed 2018-02-24 by pushkalkatara