siemens/ros-sharp #44: Talker structure on ROS end

arunavanag591, 2018-02-23:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions, I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hi,

I was looking at this example [here](https://github.com/siemens/ros-sharp/wiki/Info_CodeExample). 
I am wondering how should the talker look like in order to receive the published
` message.data = "Hello!";`

If I am not wrong, is this right?

```
#include "ros/ros.h"
#include "std_msgs/String.h"
#include <sstream>
int main(int argc, char **argv)
{
  ros::init(argc, argv, "my_node");
  ros::NodeHandle n; 
  ros::Publisher chatter_pub = n.advertise<std_msgs::String>("talker", 1000);
  ros::Rate loop_rate(10);
  return 0;
}
```

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2018-02-23:
Look at the comments in the example. You need no additional code on the ROS side. The following commands in your Ubuntu system will do:
``` 
roscore
rostopic echo /talker
rostopic pub /listener std_msgs/String "World!"
roslaunch rosbridge_server rosrbridge_websocket.launch
```

arunavanag591, 2018-02-23:
I have seen them. Actually, I will eventually send customized messages, so I wonder do I need to define the my string. Right now for some reason, i am able to connect to rosbridge from unity, and also my topic info shows rosbridge_websocket. However I just dont see my message getting published in the terminal. I checked using wire shark and based on rosbridge protocol i am getting the right command for advertise and publish. So it has to be some issue with my node. Hence I wonder do I need to custom define or describe my message at the ros end, like a data model with headers. For now I am just testing with Hello World from unity to ros. I could not get it. I could get from ros to unity.