siemens/ros-sharp #46: How to include additional standard types (eg Float, Int, etc)

LaneF, 2018-03-22:
I have ROS running a project and broadcasting a float, but get an error on the Unity side since the `MessageTypes.Dictionary` doesn't have mapping for it. String is mapped ( `{ "std_msgs/String", typeof(StandardString) },` ) but I do not understand how this is populated in order to correctly extend it to the other standard types.

My question is how do I extend the RosBridgeClient to support the other standard types and how are they populated?

Thanks

LaneF, 2018-03-22:
I ended up compartmentalizing the client dll by making MessageTypes abstract and building custom ones in Unity along with the custom Receivers but I'm still unclear how the data is populated and have been getting an error `subscribe: Unable to load the manifest for package ... ` when trying to subscribe to custom messages. In the same project, I do not get the error for a basic String message.

MartinBischoff, 2018-03-23:
Hi @LaneF ! Thanks for experimenting with ROS#.

Since you are not the first one asking that question, I wrote a Wiki Page about it: https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes

Please let me know if I left anything unclear.

P.S. I am curious what you are doing with the other Standard Messages. I'd appreciate if you tell us what your are working on out trying out. We have a special issue for that [here](https://github.com/siemens/ros-sharp/issues/20).

MartinBischoff, 2018-03-26:
@LaneF  what is your status with this issue?
Can you work with the suggestions on the Wiki page?
If so, please close this issue.
If you have any follow-up questions, please let me know.

closed 2018-03-26 by LaneF

LaneF, 2018-03-26:
@MartinBischoff Hey thanks for the write up, it's a useful addition to the wiki.

We wanted to know about doing this because we were having some issues with communication through to the ROS VM and needed to rule out std_msgs working but custom msgs not working. Got it sorted, turned out to be an issue in the ROS project setup and we've got it running smoothly now.

I do have some suggestions for the plugin source to make it easier to extend in Unity, I can do a pull request when it's stable, if you like.

MartinBischoff, 2018-03-27:
@LaneF Definitely I am interested in your suggested modifications of RosBridgeClient!

Please note that we split it up as is to have a generic and complete library for communicating with ROS from any .NET application, not necessarily Unity.
In Unity we keep only those scripts which contain Unity-specific extensions and particularities.

If you have a stable version, please do upload it on your GitHub fork and give me a note before doing a pull request.