siemens/ros-sharp #49: Meshes missing from tutorial turtlebot scene

hassanbot, 2018-04-18:
## I found a bug! ##

* My OS is: Windows 10 Pro, version 10.0.16299
* My Unity Version is: 2017.3.1f1
* My Ubuntu (on VirtualBox) Version is: 16.04
* My ROS Distribution is: Kinetic
* My Gazebo version is: 7.12.0, running on Ubuntu VM

### Bug description ###
Turtlebot meshes are missing from turtlebot imported through GazeboSimulationScene used in the tutorials.

### Steps to reproduce ###
1.  Start a new empty Unity project.
2.  Import `RosSharp.unitypackage` through **Assets -> Import Package -> Custom Package...**. Select all Assets for import.
3.  In the project tab, navigate to **RosSharp -> Scenes**, and double-click on **GazeboSimulationScene**.

### Observed results ###
A ground plane, a camera and a sunlight is visible, as well as a magenta rectangle.
[image]

Selecting the robot in the Hierarchy, you can see the collision geometries
[image]

Zoomed-in picture
[image]

If you navigate the Hierarchy of the turtlebot until you get to a visual with a mesh, you can see that the mesh seems to be missing
[image]

hassanbot, 2018-04-19:
Ah, I see, so the workflow should be this:
1.  Import  `RosSharp.unitypackage` through **Assets -> Import Package -> Custom Package....**. Select all Assets for import.
2.  Import the **Turtlebot2** Asset through **RosBridgeClient -> Import URDF Asset...** (requires launching the file description publisher node according to Tutorial section 2.1).
3.  In the **Project** tab, navigate to **RosSharp -> Scenes**, and double-click on **GazeboSimulationScene** to load the scene (the turtlebot added to the last scene will be gone, but a new one without meshes will be imported with the scene).
4.  Delete the **turtlebot** object from the Hierarchy.
5.  Import a new **turtlebot** object through **GameObject -> 3D Object -> URDF Model**.

However, if you do this, the new turtlebot object won't have any of the ROS publisher and subscriber components. And why is there a turtlebot object in the imported scene if you're supposed to replace it? Or am I missing something?

MartinBischoff, 2018-04-20:
No you are absolutely right. You found a flaw in our tutorials here.
Originally we did not plan to add the `.scene` files at all, but let users set up the whole scenes themselves as described in the videos.
Then we thought the pre-assembled scenes are a good reference to look at when there is anything unclear in the videos. 
But the way it is now users just wonder why the scenes are incomplete...

I see 3 options to improve that:
1. Remove the Turtlebot2 gameobject from the scene but add a tutorial-specific Turtlebot patcher which automatically completes and configures a freshly imported Turtlebot by one click in the inspector.
2. Remove the scenes and let users set up the scene themselves by following the videos.
3. Explicitly write in the tutorials that the two scenes are just for reference and work, but the robot meshes are missing.

Does anyone see another option? Which option would you prefer?

hassanbot, 2018-04-23:
I think **3** sounds good :)

MartinBischoff, 2018-04-23:
Thanks @hassanbot . Any further opinion?

MartinBischoff, 2018-04-24:
We decided for Option 3 and updated the Wiki. See Section [2.2](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_GazeboSimulationExample) and [2.3](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_UnitySimulationExample) _"Setting up the Unity scene"_.

closed 2018-04-24 by MartinBischoff

mirellameelo, 2019-02-20:
I really don't understand why you just simply share this meshes...

MartinBischoff, 2019-02-20:
@mirellameelo Please see the answer to your question [above](https://github.com/siemens/ros-sharp/issues/49#issuecomment-382404914).
* Who would you address in case you found a bug or had an improvement suggestion to the urdf, meshes or textures of the turtlebot?
* Who would be responsible in keeping the turtlebot files synced with the latest official version?

Please feel free to provide a fork of this project yourself with turtlebot meshes added.

mirellameelo, 2019-02-21:
@MartinBischoff 
Well, I'd seen your answer above, and still made me confuse, I'm sorry...
I was trying hard to make the tutorial from the beginning and I was facing some issues (I'm super noob in Unity...), so I feel kind of frustration. I did'n see the warning about the mashes... My mistake... 
But, anyway, I found a solution for now!! :) 
Tks for replying!

mirellameelo, 2019-04-25:
Hi, again!
If anyone still needs the **Turtlebot with its meshes**, I just uploaded in my [repository](https://github.com/mirellameelo/Turtlebot-Unity-URDF).  Hope it helps.