siemens/ros-sharp #53: how to realize ROS serviceServer in Unity .cs

zhangying-zdmedtech, 2018-04-27:
## I have a question! ##
**Here is my question:**

* hello 
*        I'm download ros-sharp a few days ago and flowing the Wiki to realize ROS Communication step by step. it's realy helpful. In Wiki [Info_Showcases](https://github.com/siemens/ros-sharp/issues/new) page,  here says ROS# can Subscribe and publish topics, call and advertise services, but I have not find the way use RosSocket in unity cs files to realize  ROS serviceServer ,  to be called by local ROS node.  How to realize ? 
* does it relate with rosbridge_suite?
*
* My OS is:   Ubuntu 16.04
* My Unity Version is:  Unity-2017.3.0f1
* My ROS Distribution is:  kinetic
* My Build Target Platform is:  Linux 64bit

MartinBischoff, 2018-04-30:
Hi @zhangying-zdmedtech ! Thanks for your getting started with ROS#. I am very glad to read that the Wiki is a help for you.
An implementation of a Service Server is missing in [RosSocket.cs](https://github.com/siemens/ros-sharp/blob/master/RosBridgeClient/RosSocket.cs). We simply never had the neccessity to reqest a service form the Windows side yet.
To make the RosBridgeClient interface complete (and to keep our promises on [Info_Showcases](https://github.com/siemens/ros-sharp/wiki/Info_Showcases), we will implement a proper Service Server in the following days.
You can already request a service from ROS via `RosSocket.CallService` and access parameters via the [rosapi](http://wiki.ros.org/rosapi) service provider, e.g. via
```
rosSocket.CallService("/rosapi/get_param", typeof(ParamValueString), receiveRobotName, new ParamName("/robot/name"));
```

zhangying-zdmedtech, 2018-05-02:
@MartinBischoff  Thank you!

MartinBischoff, 2018-05-03:
**Attention:
The service server extension required a type change of the Message ID from `string` to `int`.**

Background:
* According to the [RosBridge Protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/groovy-devel/ROSBRIDGE_PROTOCOL.md) ID must be of type `string`. We are correcting a design flaw with this change.
* Previously (for topic communication and service requests) only Message IDs had been passed which could be parsed as `int`.
* For Service Calls from ROS, rosbridge_suite sends a certain string Message ID looking like `"service_call:<service>:<call_no>"`. The exact same string Message ID must be sent back to ROS in the corrsponding Service Response message.  This made the wohle type correction necessary.

I hope this interface adjustment does not cause too much troubles in your ROS# applications.

MartinBischoff, 2018-05-03:
@zhangying-zdmedtech Please let me know if I left anything unclear. If not please feel free to close this issue.

zhangying-zdmedtech, 2018-05-04:
@MartinBischoff thank you for the detail explanation! it's work on my project. 
Here is a small question, in ServiceCallHandler, 
```
using Newtonsoft.Json.Linq;
using System;

public delegate bool ServiceCallHandler (JObject arguments, out JObject **reult**);
```

closed 2018-05-04 by zhangying-zdmedtech