siemens/ros-sharp #60: Blocked in tutorial 2.1

my-name-is-not, 2018-05-15:
Hi, I'm having some trouble with the Turtlebot example. I just arrived to the launch step and it gives me this error : 
`[publish_description_turtlebot2.launch] is neither a launch file in package [file_server] nor is [file_server] a launch file name`

I already try to run setup.sh and setup.shell, but i didn't help. Would you have an idea of my problem ?

By the way, there's a misprint in the turtlebot installation, it's not 'ros-kinetic-kobuki-ftd' but 'ros-kinetic-kobuki-ftdi'. So the correct line would be : 
`sudo apt-get install ros-kinetic-turtlebot ros-kinetic-turtlebot-apps ros-kinetic-turtlebot-interactions ros-kinetic-turtlebot-simulator ros-kinetic-kobuki-ftdi ros-kinetic-ar-track-alvar-msgs ros-kinetic-turtlebot-gazebo`

MartinBischoff, 2018-05-15:
Hi @my-name-is-not !
Concerning the "launch file not found error": Most likely you have not sourced your workspace yet:
In your catkin workspace type:
```
source devel/setup.bash
```
Please check out [Installing and Configuring Your ROS Environment](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment) for further info.

Thanks for the info on the package name `ros-kinetic-kobuki-ftdi`. It has been renamed only recently. We will update the wiki.

MartinBischoff, 2018-05-17:
Hi @my-name-is-not !
We updated the wiki already. If also your problem is solved, please close the issue.
Thanks!

closed 2018-05-17 by my-name-is-not