siemens/ros-sharp #69: Communicating within linux using Unity linux  build

pvnkmrksk, 2018-06-01:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

pvnkmrksk, 2018-06-01:
I have built a VR setup for insects using panda3d and ROS with many hardware components in linux for my thesis. But Panda3d is severely outdated and has limited community support and Unity is the next best thing. Is it possible to communicate between ROS and unity linux builds? It would be extremely useful if you provide hints how to do it using ROS#.

p.s. Pardon me if this was there in the wiki or is very obvious. I am new to C# and unity and assistance would be greatly appreciated.

MartinBischoff, 2018-06-04:
Hi @pvnkmrksk ! Thanks for your interest in ROS#!
Though we focused on Windows platforms in our developments, other members from this community are successfully using ROS# on Linux systems, see e.g. [here](https://github.com/siemens/ros-sharp/issues/20). We are profiting from the multi-platform support provided by Unity and .NET.

If this answers your question, please close this issue.

pvnkmrksk, 2018-06-04:
Thank you for this fantastic repo. Yes, I managed to get it all running within ubuntu by just using "ws://localhost:9090". Will post my entire repo which ported from panda3d to unity using this soon. And again, great job folks.

closed 2018-06-04 by pvnkmrksk