siemens/ros-sharp #70: How to fix the menu 'RosBridgeClient' couldn't show after import?

b2220333, 2018-06-07:
---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I try Assets→Import Package→Custom Package
and choose RosSharp.unitypackage to import.
After I import it, I can see the RosSharp Asset,  but I can't see the menu has RosBridgeClient.
1. I try run the following code and re-import again, but failed.

```
cd ~/code/Unity_project/ai2thor/Assets/
mv RosSharp RosSharp.bak
mv RosSharp.meta RosSharp.meta.bak
mv RosSharp.bak ../
mv RosSharp.bak.meta ..
```
2. I make sure I install the following packages:
sudo apt-get install ros-kinetic-turtlebot ros-kinetic-turtlebot-apps ros-kinetic-turtlebot-interactions ros-kinetic-turtlebot-simulator ros-kinetic-kobuki-ftdi ros-kinetic-ar-track-alvar-msgs ros-kinetic-turtlebot-gazebo ros-kinetic-rosbridge-*

3. I try to reboot my ubuntu16.04 and rerun Unity but failed.
My installation code is:
```
cd ~/code/download
wget https://beta.unity3d.com/download/e685bfa3c483/UnitySetup-2017.4.4f1
chmod +x UnitySetup-2017.4.4f1
./UnitySetup-2017.4.4f1

```

What can I do next?
Thank you~

MartinBischoff, 2018-06-07:
Hi @b2220333 !

1. Please check your Unity Console Window for any errors.
If one script in your Unity Project does not compile, the Editor Extensions will not appear.

2. If there are no errors, and still a custom editor extension does not appear, please make sure that the corresponding editor extension classes are in Asset subfolders named `Editor`. (This should automatically be the case if you import `RosSharp.unitpackage` as you described above.)

Further info here: https://docs.unity3d.com/Manual/SpecialFolders.html

If this solves your problem, please close the issue.

If not please send me a screenshot of your Unity Editor, with the Console Window and the Asset folder structure `Assets/RosSharp/Editor/CustomEditor` open.

I hope this helps. Have fun!

b2220333, 2018-06-08:
[image]

Am I missing something to install?
So what can I do next?
Thank you~

MartinBischoff, 2018-06-08:
In the first screenshot you have 9 errors, in the second scrrenshot you have 1 error. I can not see the menu of the Unity Editor at all. Please start an empty Unity project, import `RosSharp.unitpackage` and check if the error you are having persitsts.

b2220333, 2018-06-08:
Thank you for this hint. I will try it on my lab computer on Monday because now I could not operate that computer.

b2220333, 2018-06-11:
I try to import on a new package, but failed again.
I try to reinstall UnitySetup-2017.4.4f1, but couldn't help.
[image]

What can I do next?
Thank you~

MartinBischoff, 2018-06-11:
We can not reproduce this error.
On your screenshot I see empty error messages again and a suspicous Unity message before the first ROS# related errors.

It seems to me like this error is related to Unity or the aithor library you are using.

To  isolate the error I suggest to:
1. start a new empty Unity project
2. add an own custom menu (e.g. by adding [this](https://docs.unity3d.com/ScriptReference/MenuItem.html) script)
3. sequentially add other code and libraries you want to use in your project

In each step check the console window for errors and solve them before continuing with the next step.

[Here](https://forum.unity.com/threads/unity-2017-1-p2-issue-with-guiutility-hotcontrol.487528/) is something I found concerning the hotcontrol message in Unity.
Upgrading your Unity version to 2018.1 might also be a solution for your problems.

@ Others working with ROS# on Linux: Please support if you have an idea.

b2220333, 2018-06-11:
Should I add new C# file and run?
I found that C# file is in this format:

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;

public class NewBehaviourScript : MonoBehaviour {

    // Use this for initialization
    void Start () {
             
    }   
    
    // Update is called once per frame
    void Update () {
             
    }   
}

```

Should I replace them all with the script you provide?

Thank you~

b2220333, 2018-06-21:
Hello, after I try to install 2018.2.0b7 version and upgrade the RosSharp.unitypackage, it works!
[image]

Thank you~~

closed 2018-06-22 by MartinBischoff

goikonomou, 2022-05-03:
In my case. There were 2 versions of Newtonsoft.dll . I just deleted the one that was on assets/JsonDotNet/Standalone and Rosbrigde Client appeared as desired at the top of Unity window