siemens/ros-sharp #71: Connected but not received in tutorial 2.1

warmtub, 2018-06-08:
Hello! I am trying to connect ROS node and unity both on a same ubuntu console, and I have tried the tutorial 2.1 to spawn the turtlebot. The picture below is my result. It seems that the ROS node is connected with unity since it responds that a client is connected and disconnected, but unity receives nothing even I increase timeout.

[image]

Is there anything I can try to check out what is going wrong? Thank you!!!

MartinBischoff, 2018-06-11:
Hi @warmtub !

I all looks correct on the console window on the right.

The messages in the Unity UI are misleading:
1 and 5 only states that it _tried_ to connect and disconnect. Not that this action has been performed successfully.

So I presume there is something wrong with the URI `ws://127.0.0.1:9090` that you specified.

A timeout of 10 seconds will be absolutely sufficient to receive all Turtlebot 2 data.

In issue #69 @pvnkmrksk informed us that he sucessfully up a localhost connection between ROS and Unity. @pvnkmrksk can you support @warmtub ?

pvnkmrksk, 2018-06-11:
This is what I used. I attached this to gameobject (main camera in my case)and ran it. I changed nothing inside unity. I modified the [https://github.com/siemens/ros-sharp/wiki/Info_CodeExample](url) code very little to setup the rudimentary connection. The url and the inheritance of MonoBehaviour. Do try to get this running before a turtlebot is all I can suggest.

using System;
using RosSharp.RosBridgeClient;

// commands on ROS system:
// roslaunch rosbridge_server rosbridge_websocket.launch
// rostopic echo /talker
// rostopic pub /listener std_msgs/String "World!"

public class HelloWorld: MonoBehaviour

{
    public static void Main(string[] args)
    {
        RosSocket rosSocket = new RosSocket("ws://localhost:9090");

        // Publication:
        string publication_id = rosSocket.Advertise("/talker", "std_msgs/String");

        StandardString message = new StandardString();
        message.data = "Hello!";
        rosSocket.Publish(publication_id, message);

        // Subscription:
        string subscription_id = rosSocket.Subscribe("/listener", "std_msgs/String", subscriptionHandler);

        Console.WriteLine("Press any key to close...");
        Console.ReadKey(true);
        rosSocket.Close();
    }
    private static void subscriptionHandler(Message message)
    {
        StandardString standardString = (StandardString)message;
        Console.WriteLine(standardString.data);
    }
}

warmtub, 2018-06-11:
Hello @MartinBischoff . Thanks for your explanation!

Hello @pvnkmrksk . Thanks for your suggestion! I create a new empty project to test connection. I import the RosSharp unity package and adding the script you provided on the Main Camera.
After playing the unity game and running those three ROS commands in the comment, there is nothing showing up in both side... Is there any step I missed?
[image]

MartinBischoff, 2018-06-22:
Your Main function will never called in Unity. Call it `Start()`.
Before proceeding I suggest watching some Unity scripting beginner tutorials [here](https://unity3d.com/de/learn/tutorials/s/scripting) and looking at our reference scripts [here](https://github.com/siemens/ros-sharp/tree/master/Unity3D/Assets/RosSharp/Scripts/MessageHandling). Also reading the [Wiki](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes) will help.

warmtub, 2018-06-25:
I succeed in running tutorial 2.1 on another computer. Maybe I will develop on it and skip this problem.
Thanks for your help!

closed 2018-06-25 by warmtub