siemens/ros-sharp #72: Running rosbag commands from C#

pvnkmrksk, 2018-06-08:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

pvnkmrksk, 2018-06-08:
Hey there. I have managed to make ros and unity  linux talk to each other within a single system. All thanks to rossharp. Now I need to record some bags. I intend call these from unity based on some keypresses. I tried process from system diagnostics. Never even managed to start rqt. Tried running a separate bash script too, no luck. I guess this should be quite straightforward to call rosbag command from within unity, I am just unable to figure it out. Any help would be appreciated.

MartinBischoff, 2018-06-11:
Hi @pvnkmrksk ! To my best knowledge, you are the first one interested in running rosbag commands from within Unity/C#.

The ROS# interface to ROS is nothing but a rosbridge_suite client implementation.
Thus, if [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) has the possibly to record bags, start rtq or whatever, you can directly do so via ROS#/RosBridgeClient.

If [rosbridge_suite/rosapi](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosapi) does not provide the required functionalities, you will have to set up a separate service on your ROS system, which does what you want (start rqt, etc.)
You then can call this service via rosbridge_suite as any other rosbridge communication.

You can take e.g. the [file_server](https://github.com/siemens/ros-sharp/blob/master/ROS/file_server/src/file_server.cpp) service as a reference which provides the functionality to send file contents via rosbridge_suite.

pvnkmrksk, 2018-06-11:
I see, that makes sense. I am just starting C# and Unity, so implementing myself seems beyond my reach for now. But if I change my request, running rosbag commands somehow. In python, I was using subprocess.Popen to call rosbag command line utilities to start and stop bags. I tried to mimic that using system.diagnostics.process. I just couldn't get it to run. Then I tried calling a python script using system dignostics which internally called rosbag, failed too. I just need to call a standard ros utility from unity and that would be great. A C# api implementation is not necessary for the purpose. 

But it is surprising no one was interested in recording the data streams? Wouldn't recording be the next step after communication, if not as data but for posterity reasons?

MartinBischoff, 2018-06-22:
Likely I did not really understand what you want to do.
Of course you can record all data you have on Unity and ROS side with all means you have.

Why don't you simply run your rosbag tool on your ROS system and communicate the data you want to store in your bags from Unity as common topics?

MartinBischoff, 2018-07-24:
Hi @pvnkmrksk what is the status of your issue?

As suggested above you can wrap any command you want to execute on your ubuntu system in a ROS service as explained [here](http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28c%2B%2B%29) and than call that service from Windows / Unity as shown in [this](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExampleTestProtocol.cs) example script.

Further could you please explain on your use case? Why you actually want to run those scripts from remote/unity and can not call rosbag scripts for message recoding directly at startup of your ROS system?

pvnkmrksk, 2018-07-24:
Snap. I forgot to close this, my apologies.  I run this in the start method. Initially it never ran any ROS related commands, ran everything else well. Then I realized I had to start Unity using
 '/opt/Unity/Editor/Unity %U ' from the terminal. This loaded all the ros source commands from bash which let me run any ros commandline tools like rosbag, rqt_view etc. 

I am not sure running Unity from the terminal was obvious to others. But doing that fixed everything, as standard C# code to run new proc works sweet. If anyone has similar issues, feel free to contact me. 

And thank you for your assistance. 

`           string command = "rqt";
	     string argss = "";
            System.Diagnostics.ProcessStartInfo procInfo = new System.Diagnostics.ProcessStartInfo();
            procInfo.WindowStyle = System.Diagnostics.ProcessWindowStyle.Normal;
            procInfo.FileName = command; // 'sh' for bash
            procInfo.Arguments = argss; // The Script name
            System.Diagnostics.Process p= System.Diagnostics.Process.Start(procInfo); // Start that process.
`

closed 2018-07-24 by pvnkmrksk