siemens/ros-sharp #74: ROS Bridge Client Error

MuratCA77, 2018-06-15:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I have installed Unity3D in my Linux machine and ROS-Unity3D seems to be communicating properly. I am experimenting Unity Simulation using TurtleBot and ROS seems to be controlling the robot. I would like to know what this error means in that attached screenshot of my simulation. Am I getting this error because I am not using Windows platform since my understanding is that ROS Bridge Client runs on the Windows platform?
[image]

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu
* My Unity Version is: 2018.2.0b7 Linux Experimental
* My ROS Distribution is: ROS Kinetic
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2018-06-19:
Hi @MuratCA77 ! Have you found the reason for this error yourself in the meantime?

You need to add (drag&drop) a MessageProvider to your UnityTimePublisher to get rid of this error.

Please take a look at the example scenes [UnitySimulationScene](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scenes/UnitySimulationScene.unity) and [GazeboSimulationScene](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scenes/GazeboSimulationScene.unity) for reference.

When there is an error next time, before posting an issue,  please check out the console debug message thoroughly. Take a look at the corresponding lines of code and google for the error message.
`NullReferenceException` for example is explained [here](https://docs.unity3d.com/Manual/NullReferenceException.html).

If this answers your question, please close this issue.

closed 2018-06-20 by MuratCA77

Xunzhaocunzi, 2018-12-06:
Hi MuratCA77, can you explain a little bit how you made the ros# work in Ubuntu? step by step would be great. I spent a lot of time trying to make it work, but failed. Thank you.