siemens/ros-sharp #75: Messages not being received

faizan-m, 2018-06-19:
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Hi!

I am trying to get the LaserScan message in using the instructions given in adding new message types. However, the messages are not being received at all as opposed to issue #37 where the messages were null. The Subscriber.cs file never calls the receive function for that topic. Things on ROS end seem to be fine. Any ideas how might I go about debugging this?

Thanks!

MartinBischoff, 2018-06-22:
Hi @faizan-m !
Thanks for testing the code on my fork! It is not ready yet but under development. Otherwise you would find it here. If you want to accelerate the development please feel free and do so.

MartinBischoff, 2018-06-22:
Let us discuss the developments on my fork further in [this issue](https://github.com/siemens/ros-sharp/issues/59).

closed 2018-06-22 by MartinBischoff

faizan-m, 2018-06-25:
Thanks for the reply!

I was trying the code from your fork in the hope that it would resolve the problem I am getting in this branch too. I have explained it in detail in #78 now!