siemens/ros-sharp #76: Cannot receive image on the image receiver in Unity when using a sphere to project the image on

stevensu1838, 2018-06-21:
* My OS is: Ubuntu14.04 
* My Unity Version is: Unity2017.4.1
* My ROS Distribution is: Indigo 
* My Build Target Platform is: 

My question is 
Cannot receive  the image on the image receiver in Unity when using a sphere(not a plane as it is instructed in the video tutorial) to project the image on
The linked screenshot is the video I followed in your tutorial and the only difference I had is that I used a sphere not a place to project the image on. Can you please tell me is it doable with a sphere? 
[https://drive.google.com/file/d/1Hch9-QDd1oIG6GGeIkLqwlknR2cH6Klc/view?usp=sharing](url)

Also, in the tutorial 2.2 Gazebo Simulation Example
Topics to be subscribed by Unity from ROS using ROS#:

1. /odom
2. /joint_states
3. /camera/rgb/image_raw/compressed

Is there any chance I can use other topics here? I wanted to charge  topic /camera/rgb/image_raw/compressed to other camera topics to receive colourful image or point cloud image but I didn't make it. Can you please give me some instruction? Thanks a lot.

MartinBischoff, 2018-06-22:
Hi @stevensu1838 

Unfortunately the link you provided does not work on my side.

**Your 1st question:**
The CompressedImageData you receive in Unity is converted to a common [Texture2D](https://docs.unity3d.com/ScriptReference/Texture2D.html) Unity object (see [here](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/MessageHandling/ImageReceiver.cs).
This Texture2D object can be applied on any mesh you like.

Do you not receive the image data? ...or can you not map the texture that you have onto the sphere?
You can identify the difference easily by placing a debug.log message into the ReceiveMessage method of your ImageSubscriber.
If you receive it but can not map it, please note that your question is better asked in a Unity forum.

**Your 2nd question:**
The intention of ROS#/RosBridgeClient is to provide a Client for [rosbridge suite](http://wiki.ros.org/rosbridge_suite) in Unity. So _absolutely_ there is a chance to use other topics here. The application examples provided in the wiki are nothing but applicatin examples.
Please note that if your camera image in ROS is not of the type [sensor_msgs/CompressedImage](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/CompressedImage.html) you will have to write your own message type as explained [here](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes).

MartinBischoff, 2018-06-25:
Hi @stevensu1838 !
Any updates on  your issue? Did you manage to resolve it in the meantime ? What was the reason?

stevensu1838, 2018-12-05:
ROS sharp release 1.3 
Purpose: change the image on image receiver to a colour one 
Topics to be subscribed by Unity from ROS using ROS#:
Change this topic /camera/rgb/image_raw/compressed (**sensor_msgs/CompressedImage**) to /camera/rgb/image_raw(**sensor_msgs/Image**)

I've noticed that both sensor_msgs/CompressedImage and sensor_msgs/Image are build in message type. So I don't think I need to create my own type of message. 

What I did:  change all **SensorCompressedImage** to **SensorImage** in order to receive topic /camera/rgb/image_raw
```
using System;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
    [RequireComponent(typeof(MeshRenderer))]
    public class ImageReceiver : MessageReceiver
    {
        public override Type MessageType { get { return (typeof(SensorImage)); } }

        private byte[] imageData;
        private bool isMessageReceived;

        private MeshRenderer meshRenderer;
        private Texture2D texture2D;

        private void Awake()
        {
            MessageReception += ReceiveMessage;
        }
        private void Start()
        {
            texture2D = new Texture2D(1,1);
            meshRenderer = GetComponent<MeshRenderer>();
            meshRenderer.material = new Material(Shader.Find("Standard"));
        }
        private void Update()
        {
            if (isMessageReceived)
                ProcessMessage();
        }
        private void ReceiveMessage(object sender, MessageEventArgs e)
        {
            imageData = ((SensorImage)e.Message).data;
            isMessageReceived = true;
        }

        private void ProcessMessage()
        {
            texture2D.LoadImage(imageData);
            texture2D.Apply();
            meshRenderer.material.SetTexture("_MainTex", texture2D);
            isMessageReceived = false;

```
Then I rebuilt the image receiver in Unity by following the video tutorial. In rqt_graph it shows ROS is publishing ### /camera/rgb/image_raw through ROSbridge successfully. However, I don't get any image on the plane. (### It worked before when I subscribed topic /camera/rgb/image_raw/compressed )
[image]

Can you please kindly show me what I did wrong? Thanks a lot