siemens/ros-sharp #77: Urdf not appearing in Unity when imported from ROS

Ewaolx, 2018-06-25:
* My OS is: 16.04.3
* My Unity Version is: 2018.1.1
* My ROS Distribution is: Kinetic

**Here is my bug description:**
I am trying to import URDF from ROS in Unity. 
 
There are couple of issues, I am facing:
- The problem is the URDF is not visible in 'game' and 'scene' in Unity. Though in the Assets section of Unity I can notice the URDF being imported.
[image]

- The other issue seems that, the xacro file(robot_description) is not being fully imported in Unity as there are several links in xacro file, but only base_link is present in Unity.
[image]

URDF Details:
https://bitbucket.org/DataspeedInc/dbw_mkz_simulation/overview

Ewaolx, 2018-06-27:
Hi Martin,
I have attached the folder containing urdf and the mesh file.(Link below)
 [](url)https://drive.google.com/open?id=1gMY1AFkGuEjS8xznoekkflxn2q9HnEsR

- Yes, I have started the` file_server file_server.launch `. I am launching the file which in turn is launching the file_server.launch as suggested in the tutorial(2.1).

- No, I am not receiving any errors or warnings during the URDF import.

- Yes. The meshes are visible in the project tab(below in Unity), when they are imported manually through coping as well as when done from ROS.(image below)
[image]

Moreover,  when I try to run the URDF in ROS, it runs fine without any errors(gazebo image below), but in Unity it doesn't show up as described earlier.
[image]

Further resources about the simulation(packages) can be found by running two commands provided in below link.
[](url)https://bitbucket.org/DataspeedInc/dbw_mkz_simulation/overview

Thanks for sharing interest in my application. I am trying to establish communication between ROS and Unity using ROS# so that I can run the URDF(above link) in Unity simulation instead of Gazebo and have it visualize in ROS(rviz) with the help of sensor messages.

MartinBischoff, 2018-07-05:
Good news! I just fixed your urdf file and imported your nice car model into Unity:

[image]

I had to make 3 adjustments to get the model in Unity as we are seeing it. I solved the import crash you reported above with adjustment 1 only:

1. In line 18 of the urdf file I changed:
```<mesh filename="file://meshes/body.dae" scale="0.38 0.394 0.394"/>```
to:
```<mesh filename="package://dbw_mkz_description/dbw_urdf/urdf/meshes/body.dae" scale="0.38 0.394 0.394"/>```

Both `file_server` and `UrdfImporter` support `"package://<path_to_mesh_file>"` only. This is the reason why the file contents of `body.dae` was empty on Windows after importing it via rosbridge and the UrdfImporter stopped with an error at that point since it could not even find the empty file at that location.

2. Originally I had problems with the import of the following 3 dae files:
[image]
Most likely I just didn't have the corresponding ROS code to set the parameters. So in all 3 files `VLP16_base_1.dae`, `VLP16_base_2.dae` and `VLP16_scan.dae` i changed line 74 from
```<node id="<STL_BINARY>" name="<STL_BINARY>">```
to ```<node id="VLP16_base_1" name="VLP16_base_1">```,
```<node id="VLP16_base_2" name="VLP16_base_2">``` and 
```<node id="VLP16_base_scan" name="VLP16_base_scan">```.

3. In line 25 of the URDF I changed:
```<mesh filename="package://dbw_mkz_description/dbw_urdf/urdf/meshes/body_collision.stl" scale="0.01 0.01 0.01"/>```
to:
```<mesh filename="package://dbw_mkz_description/dbw_urdf/urdf/meshes/body_collision.stl" scale="0.001 0.001 0.001"/>```
It seems like the scaling of the mesh in the stl file that is used for the body collider is wrong by a factor of 10 in Unity. If this is not also the case in Gazebo, it might be a bug in the StlImporter.

MartinBischoff, 2018-07-24:
Hi @Ewaolx what is the status? Have you sucessfully imported the URDF with the fixes I suggested above? Please write a short comment on the current status and close the issue if you think it is done.

samiamlabs, 2018-07-26:
Hi.

I had some minor issues when importing URDF:s from ROS after the recent changes.
It seems like the ROS file server does not work with the "WebSocketNetProtocol", which is the default. If you switch to "WebSocketSharpProtocol" it works fine.

Se video:
[image]

I also noticed that shaders disappear from the first robot if you import a second one.

MartinBischoff, 2018-07-27:
Hi @samiamlabs !
Thank you for this report!

**UrdfImport with WebSocketNetProtocol:**
Please do update your Unity Version to 2018.x. Most likely this is the reason for this error.

URDF Import should work equally well with all protocols. You are working with Windows 10, so both protocols should work equally well on your system. We have already have sucessfully tested UrdfImport both for WebSocketNetProtocol and WebSocketSharpProtocol with multiple different robots.

**Disappearing materials:**
When importing a second time, all files are overwritten.
We want them to be overwritten, to make sure that any mesh or texture changes in the ROS / Ubuntu System are applied in Unity as well.
Do you have a better idea how to proceed in this concern?

Please open a separate issue for reporting a new technical problem next time ( though your problem also fits well to the title of this issue in this case).

Ewaolx, 2018-07-27:
Hi Martin,
I haven't actually tried the fixes you suggested, as I am busy working with Unity for other modules of my application. But my application will certainly require ROS# to import this URDF. So, I am glad that you suggested this fix and will surely look into it shortly. But for now, I am closing this issue as I think the fix will solve the issue.

Thanks.

closed 2018-07-27 by Ewaolx