siemens/ros-sharp #78: Bug in ROSBridgeClient?

faizan-m, 2018-06-25:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I found a bug! ##
* [ x] I am using the latest ROS# version available here on the master branch.
* [x ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**
I am trying to get the Laser Scan topic inside Unity. It subscribes well but the callback function is never called so data never comes. 

Perform the following steps reproduce the bug:

I followed the instructions [here](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes):
Added the SensorLaserScan class in Message.cs:
`    public class SensorLaserScan : Message
    {
        public StandardHeader header;
        public float angle_min;
        public float angle_max;
        public float angle_increment;
        public float time_increment;
        public float scan_time;
        public float range_min;
        public float range_max;
        public float[] ranges;
        public float[] intensities;

        public SensorLaserScan()
        {
            header = new StandardHeader();
            angle_min = 0;
            angle_max = 0;
            angle_increment = 0;
            time_increment = 0;
            scan_time = 0;
            range_min = 0;
            range_max = 0;
            ranges = new float[0];
            intensities = new float[0];
        }
    }`

Added the corresponding values in the MessageTypes.cs:

`            { "sensor_msgs/LaserScan", typeof(SensorLaserScan) },
`

Created a Test Script in Unity:

`using System;
using System.Collections;
using System.Collections.Generic;
using UnityEngine;

using RosSharp.RosBridgeClient;

public class TestScript : MonoBehaviour {
    RosSocket rosSocket;

    // Use this for initialization
    void Start () {
        Debug.Log("Start Called on Script");

        rosSocket = new RosSocket("ws://172.16.218.131:9090");
        string subscription_id = rosSocket.Subscribe("/scan", "sensor_msgs/LaserScan", subscriptionHandler);
        Debug.Log("Subscription ID: " + subscription_id);

    }

    // Update is called once per frame
    void Update () {
        Debug.Log("Update Called on Script");
    }

    private static void subscriptionHandler(Message message)
    {
        Debug.Log("Callback Called on Script");

        SensorLaserScan scanData = (SensorLaserScan)message;
        Debug.Log(scanData.angle_max);
    }
}
`
Observed results:
Callback function is never called, though a subscription id is given and rosbridge_suite acknowledges that it has subscribed.

Expected Results:
Used this exact code for a CompressedImage topic and it works as expected, calling the callback function as the data is available.

MartinBischoff, 2018-06-25:
I throughly checked the code you provided above. It all looks correct.

In another not publicly available application we already communicated LaserScan messages from ROS to Unity via this ROS# version and by nothing but the straight-forward additions to Messages and MessageTypes that you also did above.

The only ideas I have are:
* Are you actually publishing messages on the `/scan` topic on your ROS system? I.e. is there output when you type `rostopic echo /scan`?
* Have you updated the RosBridgeClient.dll in your Unity project? Does this contain the code you sent me above?

You can debug your code outside of Unity by modifing [this](https://github.com/siemens/ros-sharp/wiki/Info_CodeExample) console application in your Visual Studio project. Can you receive your LaserScan messages there?

faizan-m, 2018-06-25:
Thanks for the reply! I found the bug!

The /scan topic gives nan to values that it did not get a value for by laser scan. This nan is converted to null in json and it causes the bug in the receivedPublish function in RosSocket.cs line 255 specifically in the bit where the call is made to ToObject function. 

Instead of assigning a nan to the float or even 0 as a default to that particular value, it just straight up fails and discards the whole message so the call is never made to the subscriber function.

Hence as it stands, ROS Sharp will reject the whole message if there is a single value in there somewhere that is nan. Took me about to a week figure this all out but hopefully this can be fixed in future versions!

MartinBischoff, 2018-06-26:
@faizan-m : So how would you like to have the line 255 changed in future versions then?

faizan-m, 2018-06-26:
I would expect it to behave a little more predictably. For values that are null, just use default constructor values for example or throw an exception so user knows what happened.

But I am not sure about the exact changes that need to be made because I don't really know the Json plugin very well. Maybe the solution is to modify/check the message before converting to object to remove nulls? Or maybe it is unavoidable and you can make a note of it in the documentation somewhere?

MartinBischoff, 2018-06-26:
[Here](https://www.newtonsoft.com/json/help/html/SerializationErrorHandling.htm) is the JSon Serialization Error Handling documentation with a good example.

I would say that the correct handling of null values in messages is application specific. It depends on the hardware sensor, the message type, the ros node implementation, and the Unity code how Newtonsoft.Json and ROS# should cope with a null value. 0, NaN or discarding the whole message can be the right reaction.

For example I just remembered that for our Laser Scanner we completely avoided the null values on the Unity side by republishing the topic in ROS where all null values are replaced by 0.

If we give out a Debug.LogWarning instead of ignoring null values in the ToObject method, will will improve ROS# and make it more user friendly.

Ideally, these JSonSerializerSettings can be provided by a virtual method that can be overriden by the user.

faizan-m, 2018-06-26:
Yeah that sounds good!

berkayalpcakal, 2018-07-31:
Hello @faizan-m 

I have experienced the similar problem as you stated [here](https://github.com/siemens/ros-sharp/issues/78#issuecomment-400029439) before. 

I just added a node to ROS, which subscribes to /scan and publishes the corrected messages to another topic /scanCorrected after converting NaN values to particular values. Then from Unity, I subscribe to corrected messages on /scanCorrected. 

After a while, NaN values in the original messages just disappeared. I have not been using the scan corrector node since then. Is your sensor software up to date? 

Here I attach the script I used to create scan corrector node. This solution is implemented on ROS side, and is independent from ROS#.  More robust solution would be using Json Serialization Error Handling as suggested [here](https://github.com/siemens/ros-sharp/issues/78#issuecomment-400336753) .

[laserScanCorrector.zip](https://github.com/siemens/ros-sharp/files/2246242/laserScanCorrector.zip)

faizan-m, 2018-07-31:
Hi @berkayalpcakal!

Yeah I did the same thing to  make it work but my scanner still gives NaN once in a while so for me it is just safer to ensure that a null is never transmitted through json. So for now this issue itself is closed but this thread is open as a reminder for future enhancement to add error handling!

akirayou, 2019-01-24:
I got some problem. And I found another solution.
just add "?" after "float".  float? is nullable type of float in c#.

Like this. (Of course you have to rewrite youre subscriber code in Unity)
https://gist.github.com/akirayou/0990cd83043991a476b2a6f9d1f60ec9

It works fine... But there is another problem for me.
JSON can handle  "NaN" also.
So there is yet anather solution, Rewrite "ros_bridge" to use "NaN" insisted of "null".
Which is better do you think?

FYI:about float?
float? a;  // is not compatible with float.
a.Value ; //is float value of a
a.HasValue // is check for null

berkayalpcakal, 2020-05-14:
With [this merge](https://github.com/RobotWebTools/rosbridge_suite/pull/116) in the rosbridge_suite repository, all the NaN of Inf values are mapped to null values before serializing the messages. This is a limitation of the use of JSON as a communication element. 

Currently ROS# supports two JSON de-/serializers, see [here](https://github.com/siemens/ros-sharp/wiki/Dev_Serializers). As already @akirayou suggested, converting message class variables to nullable variables allows both of these serializers to work with JSON data. 

I do not think that rewriting rosbridge or ros# is necessary, because handling the null/NaN/Inf values is application specific.

closed 2020-05-14 by berkayalpcakal