siemens/ros-sharp #79: How to visualize data from rviz to Unity?

lee4138, 2018-06-27:
Issue Template © Siemens AG, 2017-2018
Author: Dr. Martin Bischoff (martin.bischoff@siemens.com)
-->

* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Hi all, 
I am completely new to Ros and Unity and was searching for a way to visualize rviz sensor data like laser scan or mapping data in Unity to deploy to the Hololens. Fortunately I found ROS# which makes that possible. I would like to ask which are the approaches to visualize data from rviz especially the laser scan data. I would subscribe to the topic /base_scan but obviously I have to find a way to convert the data from that topic to a format to appropriately visualize in Unity (so that the Unity scene looks like Rviz). I saw the video 'ROS#: Robot Teleoperation via Unity ' - (https://www.youtube.com/watch?v=OytzagQirrk)  from you and would like to achieve something similar. Is there any code available because in the wiki there is no part explaining that demo.  I hope for your reply and many thanks in advance!
Best regards

faizan-m, 2018-06-27:
Hi @lee4138!

I am actually in the process of creating an Augmented Reality Framework in Unity for ROS that would ideally be implementable on any devices supported by Unity, including Hololens. Unfortunately, it is still under development at the moment and not public.

I did not find any easy shortcuts to recreate rviz in Unity so I am taking the slow and steady approach of visualizing them myself.

lee4138, 2018-06-27:
Hi @MartinBischoff and @faizan-m,
 
many thanks for your reply! So with slow and steady way you mean to convert the raw data into something that will fit into unity? If you have already implemented some converter code would you mind to share please?

I watched this video 'HoloLens + ROS + AR' (https://www.youtube.com/watch?v=TkorsuwkU1g), which shows a laserscan visualization similar to rviz into the hololens. 

Unfortunately the author didnt reply to my request yet. Do you have any idea which approach he might have used? And do you have achieved something similar? 
Hope for your reply. Thanks!

faizan-m, 2018-06-27:
Yeah I am just creating graphics based off the values through ROS#. I don't really have any "converter code" in between, everything is wired up very directly the way I am doing it so there are no discrete pieces that might be useful to you at the moment.

I am not sure about the approach in the video but it has a lot of latency that I have not experienced.

lee4138, 2018-06-28:
okay, thanks for your answer! I will try that too and report.

lee4138, 2018-07-09:
Hi all,

i have now build up the connection and can receive PointCloud2 messages which I converted from the Laser Scan data. For that I have build a SensorReceiver.cs file and attached it to the Subscriber Component of the RosConnector, like done in the tutorials with the JoyReceiver. I extracted x y z values 
out of the PointCloud2 messages. 

I would like to ask what you would suggest for the next steps to visualize the data correctly in Unity. My idea is to use these xyz points to project circles into the Unity scene at that points but I am not sure about few things. I am not sure what the xyz values represent exactly, they have values ranging from (-5 to 2.5) someone have experince in working with PointClouds in Unity?

I think that I also need to do some kind of a coodrinate transformation from the RVIZ map to the Unity map in order to project the circles at the right spot. How can I do that? And where do I have to attach the scripts ? (sorry if the questions might seem basic , I am a complete beginner) .

MartinBischoff, 2018-07-28:
Hi @lee4138 ! 
* You will have to do a coordinate transform. Unity is left-handed, Y-up. All ROS applications are right-handed Z-up.
The scripts Ros2Unity and Unity2Ros in [this class](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/Extensions/TransformExtensions.cs) might be of help for you.
* I suggest not to instantiate / update individual Gameobjects per 3d point. That will cost forbiddingly much resources.
* instead update the vertices of one (or a set of) Mesh(es) in Unity with your 3d point coordinates. Here is a nice [HowTo](https://docs.unity3d.com/ScriptReference/Mesh.html). Look for a simple high-performance shader.

Though 3D point cloud illustrations are really fun, please understand that I can't take the time to  give further application-specific coding support.

For new technical problems please open a new issue in the future.

lee4138, 2018-07-31:
Dear @MartinBischoff, 
many thanks for your precious tipss! I will try them!

closed 2018-07-31 by lee4138

kosenhitatchi, 2018-09-05:
@lee4138 Great initiative :D
I'm actually looking forward to display the point cloud of a RGBD kinect-like cam in unity for VR purpose.
How far are you with your implementation? Is it public?

Sincerely

lee4138, 2018-09-05:
Hi @kosenhitatchi,
I ve implemented a visualization method which is working although not as efficient yet. It is not public yet but I will publish this as soon as some clean up is done and comments are added (latest till next week). Then I will inform you! :)

lee4138, 2018-09-09:
Hi @kosenhitatchi, 
I uploaded my Implementation for visualizing the laserscanner on my fork under PCL2 Handling.rar .
I couldnt find time to comment all and tidy it up though. Hope it helps somehow!

kosenhitatchi, 2018-09-10:
@lee4138 , what a wonderful new. I'll try your code right away and give you some feedback as soon as I can.

kosenhitatchi, 2018-09-10:
--I'm sorry in advance if I'm spaming this thread a little bit--
@lee4138  I'm actually in the process of updating your code to work with the code refactorisation (see #59).

Could you  tell me what each of your classes are doing. 
I do not understand what _SP2.cs_ is supposed to do.? thanks

akashcastelino, 2018-12-14:
Hey @kosenhitatchi @lee4138 were you all able to visualise a pointcloud using ros sharp ?

I have used two subscribers that subscribe to the RGB and depth images from a realsense d435 and use the constructor in pointcloud.cs to create an object with the pointcloud data, but I'm receiving 'NullReferenceException: Object reference not set to an instance of an object'.

If you ll can share the work youll have done with the pointcloud visualision in unity and or any help would be appreciated.

kosenhitatchi, 2018-12-14:
Good evening @akashcastelino,
I recently had to pause my development on this topic due to external factor, thus I did not advance much on this issue.
Keep me updated though :-)

AlbertoGhiotto, 2019-02-08:
Hi @lee4138, @akashcastelino 
I'm trying to visualize a point cloud map constructed using [rtabmap](http://wiki.ros.org/rtabmap_ros) which publishes this message: 
topic:  /rtabmap/mapData     
type:  rtabmap_ros/MapData
I successfully connected a linux pc running ROS and rviz which is visualizing the map with another pc running unity using [this tutorial](http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge) but I don't actually know how to visualize the point cloud map on unity. 
Do I have to write a new c# script? If so where can I find some guidelines to do so?
Thanks

stevensu1838, 2019-02-12:
Hi @AlbertoGhiotto,
I am afraid you won't get any feedback on this topic from these experts here. I totally understand your problems and I tried it before. But it seems you have to be an expert of Unity to get the job done and no instructions online at all. Please just google and take a look at this thesis.

> Visualization of Robotic Sensor Data with Augmented Reality

 
Hopefully, you can make it.  
Also, I sincerely hope the experts here can give some guidance on this topic. Coz it is really hard for our non-experts to really do something with ROS#.

lee4138, 2019-02-12:
Hi all, 
For visualization you need additional scripts that work with the incomming ROS data. 

I managed to visualize the pointclouds within the Unity Application and Hololens using different scripts and classes. Please refer to my code that I uploaded (as zip file) and posted in an answer above. There is also an explanation to that in the answers. I will try to make a git for the project of mine but dont know when exactly :/ 
Hope this helps and if there are specific questions to the code please dont hesitate to ask, when I have time I will answer! 
Best regards

stevensu1838, 2019-08-23:
@MartinBischoff 
Hi Martin, 
In terms of updating the vertices of one (or a set of) Mesh(es), you once mentioned we could update 20 % of the whole mesh. I've read through all the Unity document and it says you have to do mesh.clear( ) before updating. It deletes all you meshes and vertices. Can you please give us some hints how you update 20% of your meshes? Thanks a lot.

Dkrhub, 2020-10-18:
@linhdoan8   Hello, I am doing the same job as you recently, but I have no idea, based on your work, to visualize the point cloud in unity, can provide me with the code for reference, I would be very grateful

Janebek, 2021-09-09:
where is your code ?