siemens/ros-sharp #82: LaserScanWriter and UrdfImportTester missing from repo

dwhit, 2018-07-24:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: 2017.4.7.1f1 LTS
* My ROS Distribution is: ROS Indigo
* My Build Target Platform is: Windows Standalone

**Here is my bug description:**

There are two missing files from the current build.  

Perform the following steps reproduce the bug:

When I first open the project, I get the following error:

`Assets/RosSharp/Scripts/RosCommuncation/LaserScanSubscriber.cs(89,38): error CS0246: The type or namespace name 'LaserScanWriter' could not be found. Are you missing an assembly reference?`

Looking in LaserScanSubscriber.cs, there is a reference to a LaserScanWriter component that does not exist in the repo. Once I comment that out I get a second error.

`Assets/RosSharp/Editor/UrdfImporterTesterEditor.cs(7,13): error CS0246: The type or namespace name 'UrdfImportTester' could not be found. Are you missing an assembly reference?`

Looking in UrdfImporterTesterEditor.cs, there is a reference to a UrdfImportTester object, that does not exist elsewhere in the project. After commenting references to UrdfImportTester out of UrdfImporterTesterEditor.cs, there are no more errors.

Observed results:
Two errors due to missing references.

Expected results:
No errors.

MartinBischoff, 2018-07-25:
Hi @dwhit ! Thank you for this report!

 I accidently merged these faulty scripts upstream. They were not supposed to be here.

I removed them in [this](https://github.com/siemens/ros-sharp/commit/9ca33760c1d9dfd8cacb7736f557a54ab00bc01c) commit.

* UrdfImportTester.cs was a test script that we used for developments on my fork.
* LaserScan-related scripts are still under development. We are going to merge them upstream when ready. Feel free to have a sneak-peek on the current state of development on [my fork](https://github.com/MartinBischoff/ros-sharp).

@dwhit Please check again and close this issue if you think it is done.

dwhit, 2018-07-25:
Looks good to me!

closed 2018-07-25 by dwhit