siemens/ros-sharp #83: Running Ros-Sharp with .NET instead of IL2CPP possible?

lee4138, 2018-07-24:
## I have a question! ##

* [x ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Hi everyone, 
I would like to integrate another project (HololensARToolkit) into ros sharp. The problem is, that the other project only works with .NET as the scripting backend (under Player Settings->Other Settings-> Configuration), while ros-sharp only seems to work with IL2CPP as the scripting backend(?).
Everytime I try to build the project with .NET following errors occur:

`UnityException: Failed to run reference rewriter with command "--target=Temp\StagingArea\Assembly-CSharp.dll" "--target=Temp\StagingArea\Unity.TextMeshPro.dll" "--target=Temp\StagingArea\UnityEngine.Timeline.dll" "--target=Temp\StagingArea\UnityEngine.UI.dll" "--target=Temp\StagingArea\System.Numerics.dll" "--target=Temp\StagingArea\UnityEngine.SpatialTracking.dll" "--target=Temp\StagingArea\UnityEngine.HoloLens.dll" "--target=Temp\StagingArea\UrdfImporter.dll" "--target=Temp\StagingArea\RosBridgeClient.dll" "--target=Temp\StagingArea\UnityEngine.Networking.dll" "--target=Temp\StagingArea\Newtonsoft.Json.dll" "--target=Temp\StagingArea\MathNet.Numerics.dll" "--target=Temp\StagingArea\websocket-sharp.dll`

When i switch to IL2CPP as scripting backend then ros-sharp(without the other project) is working fine.
It seems that when using .NET it cannot find the Pluggins. But I dont know why yet. Is there a way to fix this, so that ros-sharp can be build with .NET? 
Hope for help and many thanks in advance!

Best regards

MartinBischoff, 2018-07-25:
Hi @Iee4138 !
Here are the suggested player settings for ROS#:
[image]
IL2CPP is not required and not intended to be used.

The [RosSharp Assets](https://github.com/siemens/ros-sharp/tree/master/Unity3D/Assets/RosSharp) now require `.NET 4.x Equivalent` since  we built the plugins [RosBridgeClient](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/RosBridgeClient.dll) and [UrdfImporter](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/UrdfImporter.dll) using target framework .NET 4.6.

Please note that we are now very flexible in this concern:
* We used .NET 4.6 since this is the highest .NET version that works with the current official Unity release (also in Debug Mode).
* We require .NET 4.5 or higher for [WebSocketNetProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs) only. Removing this file and using another IProtocol implementation will give the opportunity to compile with a drastically lower .NET framework version.
* Community reports that  [websocket-sharp](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/websocket-sharp.dll) does not work well with UWP and .NET Standard Versions. Removing [WebSocketSharpProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketSharpProtocol.cs) and  [websocket-sharp.dll](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/websocket-sharp.dll) and using [WebSocketNetProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs) might be required for UWP/Hololens.
* When using RosBridgeClient and UrdfImporter outside of Unity, higher .NET Versions and also more modern platforms like .NET Standard should be feasible. **Confirmation needed.**

lee4138, 2018-07-25:
Dear @MartinBischoff ,

thanks for your response! Did you build a PC,MAC, standalone app? Because, unfortunately Mono is not available as scripting backend, when building UWP Applications, only .NET and IL2CPP are (if you choose UWP in the Build setting window).
It seems that the DLLs Newtonsoft.Json, MathNet.Numerics, etc. are not usable when creating UWP applications with .NET. 
Could you please try to see if it works for you if you build an UWP App with .NET as scripting backend? 

What I also found out is that the .NET protocoll doesnt seem to work, with UWP apps. For UWP the Windows.Networking.Sockets have to be used (https://docs.microsoft.com/de-de/windows/uwp/networking/sockets). 
I dont know exactly why there are these two kind of sockets and why NET sockets dont work with UWP. Its confusing because I thought .NET app = UWP app. But did you intend to make the socket communication between UWP/Hololens and ROS possible with the new WebSocketNET protocoll you implemented?

MartinBischoff, 2018-07-26:
Dear @lee4138 thanks for these clarifications.

Apparently ROS# is still incompatible with UWP.
Would you agree continuing this discussion in the existing issue #33 and close this one?

closed 2018-07-26 by lee4138