siemens/ros-sharp #85: Put this on the ROS wiki?

pepijndevos, 2018-07-27:
Looking at [ROS client libraries](http://wiki.ros.org/Client%20Libraries), for C# you are directed at [roscs](http://wiki.ros.org/roscs) which is *ancient*.
I only found this repo via Google.
If this library is anywhere near usable, I'd highly suggest to put it on the wiki somewhere, to help people find it.

MartinBischoff, 2018-07-27:
Hi @pepijndevos thanks for your message and these links.
Without having taken an in-depth look into [roscs](http://wiki.ros.org/roscs) it seems to be some completely different type of thing than [RosBridgeClient](https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClient) (though naming is similar):
* `RosBridgeClient` serializes ROS messages as JSON strings to be communicated to ROS applying [rosbrigde_suite](http://wiki.ros.org/rosbridge_suite).
* `roscs` runs on Ubuntu only, together with a ROS system. It provides ROS users the possibilty to write their ROS nodes in C# rather than (typically) in C++ or Python for which roscpp and rospy are used.

There is other  software out there that is more closely related to ROS#: [ROS.NET](https://github.com/uml-robotics/ROS.NET) and [Urdf Importer](https://assetstore.unity.com/packages/tools/modeling/urdf-importer-99316) to name two. We are not planning to set up a Wiki page that provides an overview on related and alternative software. Though useful, it is an additional effort to keep this list up to date.
Nevertheless, someone is wants to compose (and maintain) such a list, I am happy to support.

Does this answer make sense to you?

MartinBischoff, 2018-07-28:
When answering above I misunderstood your inital comment:
_"If this library_ (I understood __roscs__ , you meant __ROS#__) _is anywhere near usable, I'd highly suggest to put it on the wiki_ (I understood [__this ROS# Wiki__](https://github.com/siemens/ros-sharp/wiki), you meant [__ROS Wiki__](http://wiki.ros.org/de))  _somewhere, to help people find it."_

Definitely we can add a link to ROS# somewhere on the ROS Wiki.
* On the [ROS Client Libraries page](http://wiki.ros.org/Client%20Libraries) this link would not fit well though. See the differentiation in my comment above.

* Instead, a  ROS# link would fit well onto lists of e.g.
  * rosbridge client implementations
  * ROS interaction possiblities from within other OSes and applications
  * applications using the URDF file format

If anybody knows such compilations in the ROS Wiki  lists, please do propose adding a ROS# link!

MartinBischoff, 2018-08-06:
Hi @pepijndevos how shall we proceed with this issue?

I do not know and did not find a page in the ROS Wiki where ROS# might fit in. And it seems like people visiting this issue board do not know either.

MartinBischoff, 2018-08-27:
We announced the existence of ROS# in [various relevant places](https://github.com/siemens/ros-sharp/wiki/Info_Publications), were are going to make more publications in the future and we are happy for any support in spreading the word.

As we did not find a place in the ROS Wiki where ROS# might fit in I am closing this issue.

closed 2018-08-27 by MartinBischoff