siemens/ros-sharp #88: Importing multiple robots without materials disapperaing

samiamlabs, 2018-07-27:
## I have a feature request! ##

**I want to do this:**

I have a ROS project https://github.com/samiamlabs/dyno with multiple different robot platforms https://dyno-docs.readthedocs.io/en/latest/simulation/robots.html 

[image]
[image]

I want to be able to import several into the same project for multi robot simulation. When a second robot is imported via RosBridgeClient and file-server the materials are overwritten and disappear from the previously imported robot.

The robots all share the same materials.

**For being able to do it, I wish ROS# had the following feature:**

Existing materials should be updated rather than overwritten so that the object references don't change when importing a new URDF.

Alternately, there should be a checkbox in the URDF Asset Importer that says something like "import materials" or "overwrite existing materials".

MartinBischoff, 2018-07-28:
Hi @samiamlabs ! 
You are right, this is an improvement of the URDF importer.

I'd suggest the following functionality:
Whenever an asset (mesh or material) already exists, the user can choose in a dialog window between keeping / repalcing this/all asset(s).

With the current version you have to reference the materials manually again in all components that have been created in previous imports. If you import your robots multiple times, writing a dedicated patcher script will save you a lot of time.

MartinBischoff, 2018-07-31:
We decided for a less chatty version where old object references are simply not lost during a re-import.
So if you e.g. change the texture of a robot in ROS and re-import the model, all robots in Unity will have the same updated texture.

@samiamlabs : Would you please try out importing your robots using the latest version?
If it works for you, please close this issue. If not, please let me know.

suzannahsmith1, 2018-08-02:
Hi @samiamlabs,

This issue should now be resolved, as of [this commit](https://github.com/siemens/ros-sharp/commit/8273cca8337a85bf6fa09a377b903ee4362ca5f4). 

I addressed the STL import error, as well as the problem with disappearing materials after importing a second robot. Could you please test this again and see if everything is working for you? If so, please close the issue. If not, please let me know. 

Suzannah

samiamlabs, 2018-08-02:
Hi @SuzannahSmith,

Everything seems to work now, thanks!
[image]

Closing the issue.

closed 2018-08-02 by samiamlabs