siemens/ros-sharp #95: ROS Errors after click play in Unity

stevensu1838, 2018-08-02:
* My OS is: Ubuntu 14.04
* My Unity Version is: 
* My ROS Distribution is:Indigo

**Here is my bug description:**
Hi Sir, when I do roslaunch gazebo_simulation_scene gazebo_simulation_scene.launch, everything works well. However, after I click play in Unity, the Ubuntu terminal throw the following error. Can you please tell me how to fix it?
[ERROR] [WallTime: 1533191883.620413] [19.720000] [Client 0] [id: 326362909] subscribe: Tried to register topic /odom with type geometry_msgs/PoseStamped but it is already established with type nav_msgs/Odometry

[image]

dwhit, 2018-08-02:
It looks like you're trying to subscribe to a topic called odom with a type PoseStamped, but the topic odom has type nav_msgs/Odometry. So you have a mismatch between the types. My guess would be to change your subscription type to match the publishing type.

stevensu1838, 2018-08-03:
Hi Sir,
I think the following file(PoseRerciever) is what I need to modify in Unity. I am a really beginner of C#. Can you please show me which part I need to modify? Thanks a lot.

 using System;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
     public class PoseReceiver : MessageReceiver
    {
        public enum PoseMessageTypes { GeometryPoseStamped, NavigationOdometry };
        public PoseMessageTypes PoseMessageType;

        private Type type;

        private Vector3 position;
        private Quaternion rotation;
        private  bool isMessageReceived;

        public override Type MessageType { get { return type; } }

        private void Awake()
        {
            SetType();
        }

        private void Update()
        {
            if (isMessageReceived)
                ProcessMessage();
        }
        private void SetType()
        {
            if (PoseMessageType == PoseMessageTypes.GeometryPoseStamped)
            {
                type = typeof(GeometryPoseStamped);
                MessageReception += ReceiveGeometryPoseStampedMessage;
            }
            else if (PoseMessageType == PoseMessageTypes.NavigationOdometry)
            {
                type = typeof(NavigationOdometry);
                MessageReception += ReceiveNaviagtionOdometryMessage;
            }
        }

        private void ReceiveGeometryPoseStampedMessage(object sender, MessageEventArgs e)
        {
            position = GetPosition((GeometryPoseStamped)e.Message).Ros2Unity();
            rotation = GetRotation((GeometryPoseStamped)e.Message).Ros2Unity();
            isMessageReceived = true;
        }
        private void ReceiveNaviagtionOdometryMessage(object sender, MessageEventArgs e)
        {
            position = GetPosition((NavigationOdometry)e.Message).Ros2Unity();
            rotation = GetRotation((NavigationOdometry)e.Message).Ros2Unity();
            isMessageReceived = true;
        }
        private void ProcessMessage()
        {
            transform.position = position;
            transform.rotation = rotation;
        }        

        private Vector3 GetPosition(GeometryPoseStamped message)
        {
            return new Vector3(
                message.pose.position.x,
                message.pose.position.y,
                message.pose.position.z);
        }
        private Vector3 GetPosition(NavigationOdometry message)
        {
            return new Vector3(
                message.pose.pose.position.x,
                message.pose.pose.position.y,
                message.pose.pose.position.z);
        }

        private Quaternion GetRotation(GeometryPoseStamped message)
        {
            return new Quaternion(
                message.pose.orientation.x,
                message.pose.orientation.y,
                message.pose.orientation.z,
                message.pose.orientation.w);
        }

        private Quaternion GetRotation(NavigationOdometry message)
        {
            return new Quaternion(
                message.pose.pose.orientation.x,
                message.pose.pose.orientation.y,
                message.pose.pose.orientation.z,
                message.pose.pose.orientation.w);
        }
    }
}

dwhit, 2018-08-03:
You don't need to edit the code. Notice the public variable PoseMessageType, you need to set that to NavigationOdometry. You should be able to do that in the Unity Editor.

stevensu1838, 2018-08-04:
Hi Sir, 
Problem solved with your instructions. May I please ask you another question before I close this one?

I've done the Gazebo simulation with Turtlebot by following the wiki. In the turtlebot example in wiki 2.2 the following topics are used for communication:
Messages to be sent by Unity using ROS#:

/sensor_msgs/joy
Topics to be subscribed by Unity using ROS#:

/odom
/joint_states
/camera/rgb/image_raw/compressed
My question is I actually want to control a robot arm UR5 instead of a mobile robot like UR5. How did you decide to use the above topics to control turtlebot2 in Unity? How can I figure our what topics to send and receive between ROS and Unity to build the communication? I simply want the UR5 arm imported in Unity moves when I control the UR5 arm in Gazebo.
I am a really beginner of ROS and programming. Any help about this question will be much appreciated. Thank you so much

MartinBischoff, 2018-08-06:
Hi @stevensu1838 :

Please open new issues for new issues/questions/problems in the future to simplify following the discussion thread and to give other users the chance to find your issues before they ask the same question again. 

Please also make sure to thoroughly consider the questions asked the issue template when writing an issue.

As you we have already answered this question that you asked again in issue #99, I am closing this issue now.

closed 2018-08-06 by MartinBischoff