siemens/ros-sharp #99: How to figure out what topics to use in ROS#connector for robot arm applications

stevensu1838, 2018-08-04:
Hi Sir, 
I've done the Gazebo simulation with Turtlebot by following the wiki. In the turtlebot example in wiki 2.2 the following topics are used for communication:
Messages to be sent by Unity using ROS#:

/sensor_msgs/joy
Topics to be subscribed by Unity using ROS#:

/odom
/joint_states
/camera/rgb/image_raw/compressed
My question is I actually want to control a robot arm UR5 instead of a mobile robot like UR5. How did you decide to use the above topics to control turtlebot2 in Unity? How can I figure our what topics to send and receive between ROS and Unity to build the communication? I simply want the UR5 arm imported in Unity moves when I control the UR5 arm in Gazebo.
I am a really beginner of ROS and programming. Any help about this question will be much appreciated. Thank you so much.

samiamlabs, 2018-08-05:
Hi,

Are you using ros_control to move the arm in Gazebo like in the standard tutorials (http://gazebosim.org/tutorials?tut=ros_comm&cat=connect_ros)? Or are you using some other method?

I am possibly going to add an ABB arm to Unity with ros_control and MoveIt support soon. Similar to what I did here: https://www.youtube.com/watch?v=BmsKlj1p42Q (using SIGVerse, not ros-sharp).  (I have an upcoming project that involves throwing around circus artists with said ABB arm). If you have time to wait for that, you can maybe use it as an example.

As for message types, I would recommend http://docs.ros.org/melodic/api/sensor_msgs/html/msg/JointState.html
There is already some support for using that topic type to control joints in ros-sharp

A lot of this does depend on what you are trying to accomplish though...

stevensu1838, 2018-08-05:
Hi Sir,

Thank a lot for you prompt reply and sharing. I am on these linked materials right now. And my target is simply use the three command lines in the linked page to control a UR5 from Rviz. And I've successfully imported UR5 in Unity. However, The UR5 now is just lying there in Unity and I hope the UR5 arm in Unity can moves as the gazebo one moves.
I am still very curious how ROS# team decided to use the four topics for communication. Coz if you run turtlebot simulation in ROS alone, you get the following topics(I just want to learn how to select usefull topic for the use of UR5). Can you please explain for a UR5 arm or ABB arm. Which topics shall I send from ROS to Unity?

 
Topic list of normal turtlebot simulation in ROS alone and the bold texts are the ones selected to use in ROS# wiki
ysu66@mech1331:~$ rostopic list
/camera/depth/camera_info
/camera/depth/image_raw
/camera/depth/points
/camera/parameter_descriptions
/camera/parameter_updates
/camera/rgb/camera_info
/camera/rgb/image_raw
**/camera/rgb/image_raw/compressed**
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/camera/rgb/image_raw/compressedDepth/parameter_updates
/camera/rgb/image_raw/theora
/camera/rgb/image_raw/theora/parameter_descriptions
/camera/rgb/image_raw/theora/parameter_updates
/clock
/cmd_vel_mux/active
/cmd_vel_mux/input/navi
/cmd_vel_mux/input/safety_controller
/cmd_vel_mux/input/switch
/cmd_vel_mux/input/teleop
/cmd_vel_mux/parameter_descriptions
/cmd_vel_mux/parameter_updates
/depthimage_to_laserscan/parameter_descriptions
/depthimage_to_laserscan/parameter_updates
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
**/joint_states**
/laserscan_nodelet_manager/bond
/mobile_base/commands/motor_power
/mobile_base/commands/reset_odometry
/mobile_base/commands/velocity
/mobile_base/events/bumper
/mobile_base/events/cliff
/mobile_base/sensors/bumper_pointcloud
/mobile_base/sensors/core
/mobile_base/sensors/imu_data
/mobile_base_nodelet_manager/bond
**/odom**
/rosout
/rosout_agg
/scan
/statistics
/tf
/tf_static

Thanks a lot

stevensu1838, 2018-08-05:
[image]

MartinBischoff, 2018-08-06:
Hi @stevensu1838 . Plase note that you can easily add any ROS message or service type to ROS# as explained [here](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes), even custom ones.

Looking at the Turtlebot 2 topic list you sent above or at [standard_msg](http://wiki.ros.org/sensor_msgs) and [common_msg](http://wiki.ros.org/common_msgs) we see that there are many different message types, the data they contain can be used in Unity in an endless variety of ways.
We do not want to provide examples for each of them.
Instead, we included some reference message types in ROS# only. This selection can be applied to a variety of applications and for many robots, the code is simple to understand and provides a good reference for handling other message types. Users can adjust the code and extend the number of messages types to their needs and their application.
We want to keep the number and the complexity of examples low as each of them causes additional maintenance and documentation effort.
If you still feel that some reference code for a particular message type should be included into ROS# please do inform us or make a pull request.

As @samiamlabs already answered above, it depends on your application, which messages to communicate and what to do with the message data in Unity. 
Please also note that the objective of this issue board is not to teach ROS programming or answer application-specific questions. There a better places out there, e.g. [ROS answers](https://answers.ros.org/).

closed 2018-08-06 by MartinBischoff