siemens/ros-sharp #101: ServiceProvider script for Unity

samiamlabs, 2018-08-07:
## I have a question! ##

**Here is my question:**
I wanted a service provider in Unity and noticed that there was no script for that in RosCommunication in the Unity project. Is there a reason it is missing, or have you just not gotten around to adding it?

After digging around a bit in RosBridgeClient I eventually figured out how to implement it and the backend seems to work.

**ServiceProvider.cs**
```csharp
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
    [RequireComponent(typeof(RosConnector))]
    public abstract class ServiceProvider<Tin, Tout> : MonoBehaviour where Tin : Message where Tout : Message
    {
        public string serviceName;

        protected virtual void Start()
        {
            GetComponent<RosConnector>().RosSocket.AdvertiseService<Tin, Tout>(serviceName, ServiceResponseHandler);
        }

        protected abstract bool ServiceResponseHandler(Tin request, out Tout response);

    }
}
```

MartinBischoff, 2018-08-08:
Hi @samiamlabs :
We simply didn't add a ServiceProvider in Unity, since we never need one.
It makes a lot of sense to add one, though.
Thank you for providing these nice lines of code which also fit very well to the exisiting architecture of ROS#.

We shouldn't add an abstract class to ROS# without providing at least one example implementation that shows how to use it. I'm going to post one later below.
What are you using this ServiceProvider for? Does it make sense to add it as an additional reference?

Have you thought about a ServiceSubscriber in Unity as well?

MartinBischoff, 2018-08-08:
Here is a ServiceProvider example implementation:
```
using System;
using System.Collections.Generic;
using UnityEngine;
using rosapi = RosSharp.RosBridgeClient.Services.RosApi;

namespace RosSharp.RosBridgeClient
{
    public class GetParamServiceProvider : ServiceProvider<rosapi.GetParamRequest, rosapi.GetParamResponse>
    {
        Dictionary<string, string> ParamDictionary;

        private void Awake()
        {
            ParamDictionary = new Dictionary<string, string>
            {
                { "FixedTimestep", Time.fixedDeltaTime.ToString() },
                { "ApplicationVersion", Application.version },
                { "ScriptingRuntimeVersion",  UnityEditor.PlayerSettings.scriptingRuntimeVersion.ToString() },
                { "RosConnectorProtocol", GetComponent<RosConnector>().Protocol.ToString()},
                { "Parameters", " "}
            };
            ParamDictionary["Parameters"] = "{" + String.Join(";",ParamDictionary.Keys) + "}";
        }

        protected override bool ServiceResponseHandler(rosapi.GetParamRequest request, out rosapi.GetParamResponse response)
        {
            response = new rosapi.GetParamResponse();
            bool isFound = ParamDictionary.TryGetValue(request.name, out response.value);

            if (!isFound)
                response.value = "unknown parameter";

            return isFound;
        }
    }
}
```

* Any suggestions for more useful Parameters to be communicated from Unity to ROS?
* Should we add simpler (but useless) examples like AddTwoInts or a MessageRepeater instead?
* We could make the Parameters configurable by using 2 Lists instead of a Dictionary in Unity. The parameter values would however have to be manually defined by the user then.

samiamlabs, 2018-08-08:
I don't think it matters what the example does, as long as there is one.

I was using it to get an initial "world state" from unity for a parcel delivery example.
[image]

But I don't think that is a very good example since the service request is empty.

```
using RosSharp.RosBridgeClient.Services.Dyno;
using UnityEngine;
using System.Collections.Generic;

namespace RosSharp.RosBridgeClient
{
    public class DynoObjectsProvider : ServiceProvider<Services.Dyno.GetObjectsRequest, Services.Dyno.GetObjectsResponse>
    {
        public List<GameObject> objects;

        private Messages.Dyno.Object[] objectsMsg;

        protected override void Start()
        {
            base.Start();
            objectsMsg = new Messages.Dyno.Object[objects.Count];
        }

        private void Update()
        {
            for (int index = 0; index < objects.Count; index++)
            {
                objectsMsg[index] = new Messages.Dyno.Object();
                objectsMsg[index].name = objects[index].name;
                objectsMsg[index].type = "parcel";
                objectsMsg[index].pose.position = GetGeometryPoint(objects[index].transform.position.Unity2Ros());
                objectsMsg[index].pose.orientation = GetGeometryQuaternion(objects[index].transform.rotation.Unity2Ros());
            }
        }

        protected override bool ServiceResponseHandler(GetObjectsRequest request, out GetObjectsResponse response)
        { 
            response = new Services.Dyno.GetObjectsResponse(objectsMsg);
            
            return true;
        }

        private Messages.Geometry.Point GetGeometryPoint(Vector3 position)
        {
            Messages.Geometry.Point geometryPoint = new Messages.Geometry.Point();
            geometryPoint.x = position.x;
            geometryPoint.y = position.y;
            geometryPoint.z = position.z;
            return geometryPoint;
        }

        private Messages.Geometry.Quaternion GetGeometryQuaternion(Quaternion quaternion)
        {
            Messages.Geometry.Quaternion geometryQuaternion = new Messages.Geometry.Quaternion();
            geometryQuaternion.x = quaternion.x;
            geometryQuaternion.y = quaternion.y;
            geometryQuaternion.z = quaternion.z;
            geometryQuaternion.w = quaternion.w;
            return geometryQuaternion;
        }
    }
}
```

**Side note:** Will probably move the GetGeometryPoint and GetGeometryQuaternion to an extension on my branch

MartinBischoff, 2018-08-09:
I uploaded the ServiceProvider together with a very simple GetParam example in above commit.
Thanks again @samiamlabs for your suggestion and the reference code!

closed 2018-08-09 by MartinBischoff

chstoubos, 2018-09-07:
Hello everyone,

I want to request a service with a Pose message and get a string message as a response. The server side is ready to accept my request, but when i advertise my service it seems to be blank. I am using the ServiceProvider script but i guess i don't understand what to modify to make it work for my needs. 

Any suggestions? I can show you my code, but i am pretty sure i am having wrong understanding on the issue. I would appreciate any suggestions/instructions on what i need to do to make it work.

Thanks in advance.

samiamlabs, 2018-09-07:
Did you add your custom service type to https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClient/Services/RosApi and import the compiled ddl to Unity?

There is some information about how to do this here: https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes

chstoubos, 2018-09-07:
I already did that. I am not sure how to set up my custom service for the Pose message. 

Thanks for the reply.

chstoubos, 2018-09-07:
It appears that my custom service is correct so the problem must be with the unity script. Thanks a lot for the answer.