siemens/ros-sharp #102: Please update the Wiki.

SummerSunX, 2018-08-13:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I want to learn ROS now.But I find that the scripts were differece between before.So ,can you give me a new tutorial on ROS?
---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here, optionally attrach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

suzannahsmith1, 2018-08-16:
Hi @SummerSunX, 
Your issue is fairly vague and hard to understand. If you would like to learn more about ROS itself, please refer to the [ROS wiki](http://wiki.ros.org/). If you are looking to learn how to use ROS# (a way to communicate between ROS and .NET applications), please explain your issue further. Which part of the wiki have you noticed is out-dated? 

We try our best to keep the wiki up-to-date and, to the best of our knowledge, it reflects the latest updates to the ROS# framework. This is outlined in the [Recent Changes](https://github.com/siemens/ros-sharp#recent-changes) section of the Readme. If you would like to use the newest version of ROS#, please go through our wiki in its entirety, especially the updated tutorials in [section 2](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_ApplicationExamplesWithROSConnection) of the wiki. 

Please let us know if you find any discrepancies between the ROS# code base and the wiki. If not, please close this issue.

Regards,
Suzannah

SummerSunX, 2018-08-17:
Hi，
I'm sorry I didn't say it clearly.
When I imported the model into Unity, I found a script named “JointStateProvider” that wasn't in the project, as opposed to online videos.
So I was puzzled how to make the model move.If you know how to do it, can you tell me?
Thank you very much.

suzannahsmith1, 2018-08-17:
Thanks for clarifying! This is indeed a mistake on our part - we renamed "JointStateProvider" to "JointStateInputPasser" a few days ago, and have not updated the video tutorial yet. If you use JointStateInputPasser instead, you should be able to use animate the model. We will fix the video content soon.
Best,
Suzannah

MartinBischoff, 2018-08-27:
Hi @SummerSunX !

The tutorial videos [Unity Simulation Scene](https://youtu.be/Ctv4BioS1Y0) and [Gazebo Simulation Scene](https://youtu.be/oh4BIE5qKoM) correspond to the latest ROS# version from on the tip of the master branch.

All showcase videos by contrast are one-time ROS# demo applications which are not updated. They are often based on outdated versions and even include non-publicly available code.

The Unity component named `JointStateProvider` that occurs in some ROS# showcase videos is included in ROS# but has been renamed to `JointStateReader`.

If this answers your question, please close this issue. Please clarify otherwise.

Best,
Martin

MartinBischoff, 2018-09-05:
closing due to inactivity.

closed 2018-09-05 by MartinBischoff