siemens/ros-sharp #103: After creating a new msg, getting errors for some scripts in Unity

KonradBielecki, 2018-08-14:
Hello,
I am trying to include a new message in my project. For this, I am following the instruction "How to add new Message Types". 

After creating the .dll and copying it into the \Unity3D\Assets\RosSharp\Plugins\, I get this error messages for some scripts. 

[image]

Help, please.

suzannahsmith1, 2018-08-14:
Hi @KonradBielecki,
We recently completed a major refactoring of RosBridgeClient, as outlined in [this issue](https://github.com/siemens/ros-sharp/issues/59). The classes that are giving you errors here no longer exist in the new framework. In order to fix these errors, please update your code base by pulling the most recent version of ROS#, or use a version of RosBridgeClient from [release 1.2](https://github.com/siemens/ros-sharp/releases/tag/v1.2c), from before the major changes. 
If this does not fix the issue, please let us know.
Suzannah

KonradBielecki, 2018-08-22:
Hello,
and thanks for the answer. I created the .dll with my new MSG with the newest version of RosBridgeClient [(release 1.2C](https://github.com/siemens/ros-sharp/releases/tag/v1.2c) ). This works for me. 
Thank you.

closed 2018-08-22 by KonradBielecki