siemens/ros-sharp #105: Actionlib Client Capability

BenArtes, 2018-08-16:
I'm exploring options for controlling ros nodes using ros-sharp. Msgs and Services seem to already be implemented, but I'm looking to use [actionlib](http://wiki.ros.org/actionlib) to provide responses, updates, and preemption. As actionlib internally uses topics for communication, it should be supported by ros-sharp provided there are actionlib_client / actionlib_server c# libraries.

Is this currently supported by ros-sharp?

Is there currently a compatible actionlib suite? If not, are there any current development efforts you know about?

MartinBischoff, 2018-08-27:
Hi @BenArtes !

You are right, action communication is technically possible via [rosbridge_suite](http://wiki.ros.org/rosbridge_suite) (and thus also for ROS#) and [has been done](https://github.com/RobotWebTools/rosbridge_suite/issues/124) for browser applications in javascript already.

We have not implemented any acionlib server or actionlib client ourselves in C# and I am not aware of any ongoing developments in that direction.

Providing `ActionServer` and `ActionClient` parent classes for which users can write their own `SetGoal()` `SetCancel()`, `GetStatus()`, `GetResult()`, `GetFeedback()` methods, seems like a fun and interesting thing to do. It would provide a substantial extension to ROS#.

MartinBischoff, 2018-10-08:
We started working on supporting ROS Actions from within Unity.
To set it up properly, we noticed that we need to revise the way how messages are defined in RosBridgeClient.

phibla, 2018-12-12:
I'm very interesed in this feature. What is the current status or can I support you in any way?

MartinBischoff, 2019-03-04:
done in commit https://github.com/siemens/ros-sharp/commit/b5b1db955a72d284711aa456f2f110e97fb12b1f

closed 2019-03-04 by MartinBischoff