siemens/ros-sharp #106: Failed to build communication between rosbridge_websocket and /joint_states nodes

Bertrand64300, 2018-08-17:
Hi Team
I am trying to sync a UR5 robot arm between Unity and ROS through ROS#. I've created a joint state subscriber in ROSconnector as it shows in the ROS# wiki for turtlebot. 

However, in rqt_graph it appears the communication  between rosbridge_websocket and /joint_states nodes are in the wrong way. And of course, I could not sync the UR5 arm in Unity as all, even though the arm in Gazebo and Rviz at ROS side work well.

Can anyone of you kindly tell me how to successfully build up the link? Thanks a lot.

Unity side:
[image]

[image]

ROS side:(Please take a look at rosbridge_websocket node )

[image]

suzannahsmith1, 2018-08-17:
Hi @Bertrand64300,
I believe you are using an old version of ROS#, and following the [old Gazebo Simulation tutorial video](https://youtu.be/Oby36fUCg5Y). In order for us to be able to support you better, please consider updating to the newest version of ROS#. After doing that, following this [newer tutorial](https://youtu.be/oh4BIE5qKoM) should help you to sync the UR5 joint states between Gazebo and Unity.
Best, Suzannah

Bertrand64300, 2018-08-21:
Hi @suzannahsmith1, 
Thank you for your response. I tried to download the new version of ROS# but it seems that it is not working on my computer. Indeed, I'm using Unity 2017.4.1f1 for my project (cause some of the packages I need to use require this version of Unity) and I saw on your ROS wiki that "the RosSharp solution is built using .NET Framework 4.6. It requires Unity 2018.x to run".
Is there any way for me to achieve what I want with the old version of ROS#?
Cordially,
Bertrand

MartinBischoff, 2018-08-27:
Hi @Bertrand64300 , surely you can stick to an older version of ROS# and set up your ROS communication with that. The first video linked by @suzannahsmith1 above also is for the older version.

closed 2018-08-27 by MartinBischoff