siemens/ros-sharp #109: Building current HEAD on macOS and preparing the unity package

shoeffner, 2018-08-21:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.

I am not sure, but the errors are mostly namespace issues which turn up tons of generic answers for c# in general, but I can't find any that's working.

* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

#103 seems to be similar, but the answer is unsatisfying, as I am actually using the master branch and I do not add messages but instead compile the raw version.

**Here is my question:**

I am trying to build ros-sharp from source. Since #59 changed lots of things I don't want to use the release 1.2c which was from BEFORE that change, but the latest versions so that I don't have to change my (newly started) project in two weeks. So I decided to build the two dll's + the unitypackage myself. With limited success – how do I succeed? Or in other terms: How to build the package from source? This would also be good to know in case I want to make any changes.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: macOS 10.13.6 High Sierra
* My Unity Version is: 2018.2.4f1 (Personal)
* My ROS Distribution is: kinetic (Docker containers, image `FROM osrf/ros:kinetic-desktop-full`)
* My Build Target Platform is: macOS 10.13.6 High Sierra (I guess?)
* My Visual Studio Version is: VS 2017 7.6 (CE) on macOS

**Here is my bug description:**

I am not sure if it's a bug, it's more likely I am missing something during the build steps.

Also the problem is two-fold:

1. When trying to run the RosBridgeClientTest tests in VS Studio, at some point (after about 1740 Threads) I am presented with the error "exceptionservicescommon.cs not found".

2. Ignoring this and just building the DLLs and copying them (see "Steps to reproduce" below) over to the plugins directory, and then trying to load everything inside the RosSharp asset, I get the following errors:

```
Assets/RosSharp/Scripts/RosCommuncation/GetParamServiceProvider.cs(18,32): error CS0234: The type or namespace name `Services' does not exist in the namespace `RosSharp.RosBridgeClient'. Are you missing an assembly reference?

Assets/RosSharp/Scripts/RosCommuncation/GetParamServiceProvider.cs(22,84): error CS0246: The type or namespace name `rosapi' could not be found. Are you missing an assembly reference?
 
Assets/RosSharp/Scripts/RosCommuncation/ImageSubscriber.cs(21,47): error CS0246: The type or namespace name `Messages' could not be found. Are you missing an assembly reference?
 <---- SNIP SNIP ----->
```

And many more (60 total) of the same style, in basically all of the /Scripts files.
#103's answer was that many of the classes were no longer existent – but since I am using the HEAD version for everything this should be no problem, or is it?

Perform the following steps reproduce the bug:

1. Install Visual Studio and Unity: `brew cask install visual-studio unity`
2. `git clone git@github.com:siemens/ros-sharp && cd ros-sharp`
3. Import the Libraries/RosSharp.sln into Visual Studio.
4. Build release versions of RosBridgeClient and UrdfImporter individually (Building the parent project does not succeed as the tests fail as mentioned above in problem 1).
5. Inspired by [How to add new Message Types](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes), copy Libraries/RosBridgeClient/bin/Release/RosBridgeClient.dll and Libraries/UrdfImporter/bin/Release/UrdfImporter.dll to Unity3D/Assets/RosSharp/Plugins.

(Optionally tried: 6. Just to be sure, copy all files from the two Release folders over)

(Optionally tried: 7. Make all those files executable `chmod +x *`)

8. Open Unity, load the project Unity3D. The errors (see problem 2) should already appear. If not, try to run the scene.
9. Well, maybe it's because this is a package-stub, so export: Assets -> Export Package... -> All (double check that the dll files are ticked) -> Export... -> store as RosSharp.unitypackage
9.1 Import in new unity Project -- same issue.

Observed results:

1. test failure
2. namespace errors when trying to use the package

Expected results:

1. no test failure
2. no namespace errors

I haven't included any specifics about the ROS setup, but if you want I can attach a Dockerfile and a docker-compose.yml etc. – but it's not needed to reproduce this issue for me.

Relevant Code:
master branch

P.S.: Additionally Visual Studio issues a warning: 
```
/Library/Frameworks/Mono.framework/Versions/5.12.0/lib/mono/msbuild/15.0/bin/Microsoft.Common.CurrentVersion.targets(5,5): Warning MSB3276: Found conflicts between different versions of the same dependent assembly. Please set the "AutoGenerateBindingRedirects" property to true in the project file. For more information, see http://go.microsoft.com/fwlink/?LinkId=294190. (MSB3276) (RosBridgeClientTest)
```

If you have any further questions or tips, please let me know! I am excited to get this to work properly, as rviz is not really docker-friendly (though most other rqt utilities seem to work fine with XQuartz) and ros is not really macOS friendly ;-)

suzannahsmith1, 2018-08-22:
Hi @shoeffner,
your issues are most likely due to not referencing the correct .NET Framework in Unity. Please see issue #108 for details. If this does not fix your issue, please let us know.
Suzannah

shoeffner, 2018-08-22:
Thank you very much, that did the trick!

closed 2018-08-22 by shoeffner