siemens/ros-sharp #110: Texture import problem in UrdfImporter

dmrub, 2018-08-22:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: 2017.1.1f1
* My ROS Distribution is: Kinetic
* My Build Target Platform is: 

**Here is my bug description:**
I can import an URDF model but textures are often missing.

Perform the following steps reproduce the bug:

I think I found a bug in UrdfImporter.ReceiveResourceFile: https://github.com/siemens/ros-sharp/blob/36af83149a0918d1351158ee3a6382576570f742/Libraries/RosBridgeClient/UrdfImporter.cs#L149

While debugging I figured following race condition:
1. There is a single collada file which is received by function ReceiveResourceFile, i.e. RequestedResourceFiles dictionary contains single entry which is set to false when ReceiveResourceFile is called.
2. Collada file contains texture files, which are loaded in thread started here: 
https://github.com/siemens/ros-sharp/blob/36af83149a0918d1351158ee3a6382576570f742/Libraries/RosBridgeClient/UrdfImporter.cs#L154-L158
3. Race condition happens when UpdateFileRequestStatus function sets resourceFilesReceived flag to true before texture loading thread is started. Or precisely when condition `RequestedResourceFiles.Values.All(x => x == true)` is checked before in ImportColladaTextureFiles function RequestResourceFiles is called and new entry set to false is added to RequestedResourceFiles dictionary:
https://github.com/siemens/ros-sharp/blob/36af83149a0918d1351158ee3a6382576570f742/Libraries/RosBridgeClient/UrdfImporter.cs#L168-L169

Observed results:
Textures are missing

Expected results:
Textures should be there

I created a bug fix, but it uses an older version of ROS#: https://github.com/siemens/ros-sharp/commit/df13069a855329c0cfcf0a67ad02b2e214bfb983 

Relevant Code:
```C#
        private void receiveResourceFile(ServiceReceiver serviceReceiver, object serviceResponse)
        {
            byte[] fileContents = ((ParamValueByte)serviceResponse).value;
            Uri resourceFileUri = new Uri(((ParamName)serviceReceiver.ServiceParameter).name);

            bool updateFileRequestStatusAtEnd = true;
            if (isColladaFile(resourceFileUri))
            {
                Thread importResourceFilesThread = new Thread(() => {
                    importDaeTextureFiles(resourceFileUri, System.Text.Encoding.UTF8.GetString(fileContents));
                    updateFileRequestStatus(resourceFileUri); });
                importResourceFilesThread.Start();
                updateFileRequestStatusAtEnd = false;
            }
            Thread writeTextFileThread = new Thread(() => writeBinaryResponseToFile((string)serviceReceiver.HandlerParameter, fileContents));
            writeTextFileThread.Start();
            if (updateFileRequestStatusAtEnd)
                updateFileRequestStatus(resourceFileUri);
        }

```

dmrub, 2018-09-24:
I just tried to import my robot with the current version of ROS# (I used this manual: https://github.com/siemens/ros-sharp/wiki/User_Inst_Unity3DOnWindows). 
Import fails, but I need a way to create a Unity setup from latest ROS# sources and not from Unity package. How to do this ?

suzannahsmith1, 2018-09-24:
Thanks for your reply. We just updated that [wiki page](https://github.com/siemens/ros-sharp/wiki/User_Inst_Unity3DOnWindows) so that it explains how to get the latest commit of ROS# instead of the latest release. Please follow the wiki again and see if you can import your robot correctly. Let us know if it still doesn't work :)

dmrub, 2018-09-24:
I tested again import and have the same problem, without my patch I don't see a texture. Here is my patch ported to latest version of ROS#: https://github.com/dmrub/ros-sharp/commit/9e541cf80747bd117ba082ffd8c58a6a797df314

dmrub, 2018-09-25:
Your commit https://github.com/siemens/ros-sharp/commit/aa0ce2c12d889b5082fc17f45418b8d946f223ac works, I am closing this issue.

closed 2018-09-25 by dmrub