siemens/ros-sharp #113: Is the TwistSubscriber missing an assignedObject slot?

shoeffner, 2018-09-04:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best in finding the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

In the new version introduced with 34fb2a8ddd58c5f099b1e4b887a253b954808fb4, all publishers and subscribers are supposed to be attached to the ROSConnector GameObject – this is also shown e.g. in the new video for the gazebo scene where the OdometrySubscriber is added to the ROSConnector. The OdometrySubscriber has a slot `assignedObject` to reference the link/joint it controls. However, the TwistSubscriber does not have such a slot.

It is possible to attach a ROSConnector to a link and then add a TwistSubscriber as well, so that it can change the transform properly. However, I wonder if it's meant to work that way?

I will for now change the TwistSubscriber (and the PoseStampedSubscriber) so they work more like the OdometrySubscriber and the ImageSubscriber, thus having slots for the game objects and using the same ROSConnector.

An alternative would be to attach the subscribers to their respective links and have a reference to the ROSConnector, but that would need some major rewriting again.

MartinBischoff, 2018-09-05:
Thank you @shoeffner for reporting this bug and providing an elaborated reference solution in pull request #114 !

I just went trough all RosCommunication scripts myself and made sure that every Publisher and Subscriber has a corresponding reference component indicating where the data is read from or written to.

I also considered the inconsistency you mentioned in the pull request: All Publishers and Subscribers should have reference Components, no GameObjects. This improves both code and understandability.

Accordingly I also changed the OdometrySubscriber to have a public Transform instead of a public GameObject and updated the reference scene.

MartinBischoff, 2018-09-05:
@shoeffner Would you please check whether the changes done in pull request #115 work fo you?
If so, please close the issue.