siemens/ros-sharp #117: Can not publish images using WebSocketNET protocol

hassanbot, 2018-09-07:
# Bug description
It does not seem possible to publish images using the WebSocketNET protocol. Using the WebSocketSharp protocol works fine.

# Steps to reproduce
1.  Launch a websocket_sharp node on a computer with ROS.
2.  Create a Unity scene with a Camera GameObject, and another GameObject with a RosConnector and ImagePublisher component.
3. Set the Image Camera property of the ImagePublisher component to the Camera, and adjust the topic and ROSBridge URL accordingly.
4.  Set the Protocol property of the RosConnector component to "Web Socket NET".

# Observations
No image messages seems to reach the ROS node. The Unity application also crashes and an error box pops up with the header "Fatal error in GC" and the message "Unexpected mark stack overflow". Using the WebSocketSharp protocol works fine and images reaches the ROS side as expected.

MartinBischoff, 2018-09-07:
Thank you @hassanbot  for this report.
We can not reproduce the error as described above.
Please see below a screenshot of an image published from Unity via WebSocketNET that is illustrated in RViz:
[image]
It is working stable, even for very high image resolutions.

What is your OS? What is your Unity Version?
Please make sure you have Unity 2018.x and check out [this](https://github.com/siemens/ros-sharp/wiki/Dev_Protocols) Wiki Page about the applicability of Protocols on different Platforms.

Other than that, here is something I found about Unity throwing a [Fatal Error in GC](https://answers.unity.com/questions/24574/fatal-error-in-gc-getthreadcontext-failed-.html).

Please do keep us informed about any progess you make in tracing this error here.

hassanbot, 2018-09-07:
Sorry, I should have attached that information from the beginning.
OS: Windows 10, version 10.0.17134
Unity version: 2017.3.1f1
ROS version: Kinetic

Since it works for other message types (at least for PoseStamped), it doesn't seem to be a DotNet version issue, but I'll try upgrading to Unity 2018.x and see if I get better behavior.

Another thing that might be of interest is that I'm not running Unity and ROS on the same computer. The two computers are on the same wired network. but maybe it still introduces some latency that makes something pile up. I'll try and do some more investigation.

hassanbot, 2018-09-07:
The crash isn't immediate either, but it takes a few seconds. If I abort the game (press the Play button again) before the crash, I get this error message in the console (but the editor doesn't freeze):
```
ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
System.Net.Sockets.Socket.ThrowIfDisposedAndClosed () (at <3649452ae23840d49bd323a708498f78>:0)
System.Net.Sockets.Socket.Send (System.Byte[] buffer, System.Int32 offset, System.Int32 size, System.Net.Sockets.SocketFlags socketFlags, System.Net.Sockets.SocketError& errorCode) (at <3649452ae23840d49bd323a708498f78>:0)
System.Net.Sockets.Socket.Send (System.Byte[] buffer, System.Int32 offset, System.Int32 size, System.Net.Sockets.SocketFlags socketFlags) (at <3649452ae23840d49bd323a708498f78>:0)
System.Net.WebSockets.ClientWebSocket+<>c__DisplayClass28_0.<SendAsync>b__0 () (at <3649452ae23840d49bd323a708498f78>:0)
System.Threading.Tasks.Task.InnerInvoke () (at <c95265f74fdf4905bfb0d5a4b652216c>:0)
System.Threading.Tasks.Task.Execute () (at <c95265f74fdf4905bfb0d5a4b652216c>:0)
```

When publishing the PoseStamped message I'm not getting any error message when aborting.

MartinBischoff, 2018-09-10:
@hassanbot : Please update your Unity Editor to a 2018.x Version before trying out anything else.
See also [README.MD/Platform Support](https://github.com/siemens/ros-sharp).
Version 2018.1 is the first offical (not beta) release  supporting .NET 4.5 [see also here](https://blogs.unity3d.com/2018/03/28/updated-scripting-runtime-in-unity-2018-1-what-does-the-future-hold/).