siemens/ros-sharp #118: Batchmode/Playermode detection to avoid dialog windows

shoeffner, 2018-09-07:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stackoverflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have throughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I have a feature request! ##

**I want to do this:**

I am currently working on a unity player instance which serves as a ROS node offering services to dynamically add a robot and/or rigid objects to a scene. The goal is to be able to run physics simulations without having to set up unity on each machine (which is difficult due to its licensing system) and having reproducible and flexible simulation environments. Think of it as something similar to [spawning a model in gazebo](http://gazebosim.org/tutorials/?tut=ros_comm) or [adding a known object to the environment for motion planning](http://wiki.ros.org/motion_planning_environment/Tutorials/Adding%20known%20objects%20to%20the%20collision%20environment#Adding_a_known_object_to_the_environment).

The process I aim for is (all from the command line/in an automated build system):
- Create a unity project from scratch or use an existing one
- Add RosSharp to it.
- Create a RosConnector which also publishes a service.
- Compile the project into a player.
- Run the player.
- Issue a service call to let the player import the robot defined on the parameter server.
- Control the robot via ROS.
- (eventually: Issue other service calls to add rigid objects)

So far, my toy workflow for this is roughly as follows, with the last line calling a [method in my fork](https://github.com/shoeffner/ros-sharp/blob/feature/batchmode-import/Unity3D/Assets/RosSharp/Editor/CustomEditor/BatchModeRobotImporter.cs#L44-L82), which I wrote to import a robot in batchmode:

```bash
PROJECT=./AutoRosImport
ROS_PACKAGE=RosSharp.unitypackage
UNITY=/Applications/Unity/Unity.app/Contents/MacOS/Unity
# Create project
${UNITY} -batchmode -quit -createProject ${PROJECT}
# Set to .Net 4.0
sed -i '' 's/  scriptingRuntimeVersion: 0/  scriptingRuntimeVersion: 1/g' ${PROJECT}/ProjectSettings/ProjectSettings.asset
# Import RosSharp
${UNITY} -batchmode -quit -projectPath ${PROJECT} -importPackage $(pwd)/${ROS_PACKAGE}
# Import Robot
${UNITY} -batchmode -quit -projectPath ${PROJECT} -noUpm \
         -executeMethod RosSharp.UrdfImporter.BatchModeRobotImporter.ImportRobot \
         -rosAddress "ws://127.0.0.1:9090" -rosJointStateReaders -rosSubscribers -logFile
```
The import will later be replaced by a build process, and some parameters should then be passed to the player instance to start publishing the ROS service.

**For being able to do it, I wish ROS# had the following features:**

- A "silent" import feature. I already built parts of it to import a robot in batchmode into either a new scene or an existing one, which can be found on my fork (if a batchimport might be interesting for other people, I am happy to polish up a PR). It uses `Application.isBatchMode` to skip the import dialog windows (e.g. in [GenerateModelIfReady](https://github.com/shoeffner/ros-sharp/blob/feature/batchmode-import/Unity3D/Assets/RosSharp/Editor/UrdfImporterActions/RosImportHandler.cs#L112)), and in the future I am planning to use `Application.isPlaying` to skip them as well. If you like, I can create a small PR for these checks, it's just three lines of code to change.
- Currently, I am using `GameObject.Find(...)` to find the robot by its name. It would be nice if the robot factory would somehow create a reference to the robot GameObject – I don't know yet what a good solution to this is, maybe an array of robots at the RosConnector GameObject?

I am looking forward to hearing some feedback on these ideas and am happy if I can help in some way or another. If you think these might be useful additions, I will polish up the batch import into a PR and work on the service as if it could become a part of ROS#. If not, I will try building a separate package for the import service nodes. In either case, I would need the silent import feature and would love to have a better mechanism to find newly imported robots and GameObjects.

MartinBischoff, 2018-09-07:
Hi @shoeffner 

generating a complete URDF model at runtime is a major endeavour. It includes adding new dae and stl assets and textures as well. We are currently not planning to develop ROS# in that direction. A runtime that works at runtime imposes various restrictions and limitations on the import architecture and possibly on the URDF itself. When generating a scene from batch in your application as you described above, you might at least want to import the resources of the robot beforehand and only import and apply the contents of the urdf xml file to the running scene.

We are currently not planning to change the established and convenient workflow:
(1) Edit Scene: Perform URDF-Imports and adjust the models to application-specific needs
(2) Play Scene: Setup continuous ROS communication

Concerning your feature suggestions:
(1) "slient" import: Due to the workflow above, the dialog windows are really helpful. Feel free to import the /robot_description param in your application silently, by simply calling the /rosapi/get_param service at runtime.

(2) When defining an empty script e.g. UrdfModel derived from MonoBehaviour, and adjusting the URDF import to place this script in the root link, you will be able to clearly find all URDF models via
```
var UrdfModels = FindObjectsOfType<UrdfModel>();
```
We are currently working on an update of the URDF importer which includes those scripts to clearly identify all URDF models in a scene.

shoeffner, 2018-09-14:
Thank you, I will take a look into your suggestions about getting the `/robot_description` param silently!

closed 2018-09-14 by shoeffner